ARMmbed/mbed-os #11513: User callback with stdio uart?

keithmwheeler, 2019-09-18:
### Description

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

When following the examples in the RawSerial documentation, the receive callback never occurs.  

Suspect this is do to this reinitializing the stdio UART, per the comment on https://github.com/ARMmbed/mbed-os/issues/10531 regarding reintializing this uart. 

If the user should not reinitialize the stdio/debug UART, and on many targets this is the uart connected "USBTX/RX" pins, should this example be changed to indicate a different serial port?

maclobdell, 2019-09-24:
@bulislaw - any suggestion?

closed 2019-10-25 by ciarmcom

0xc0170, 2019-10-28:
@keithmwheeler Please read https://os.mbed.com/docs/mbed-os/v5.14/apis/filehandle.html, it should help you to redirect the console