ARMmbed/mbed-os #11515: InterruptIn failing to initialize

farrenv, 2019-09-18:
### Description

TARGET: EP_AGORA
MCU: NRF52840
MBED-OS version 5.12.4

I am attempting to use the InterruptIn function to monitor the accelerometer interrupt pin on the Agora target.  The InterruptIn function fails to initialize.
Initialization:
``` InterruptIn ICM20602_interrupt(PIN_NAME_INT_ACCEL);```
Serial output:
++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x20004E60, Not allowed in ISR context
Location: 0x18F5B
Error Value: 0x20004E60
Current Thread: main  Id: 0x20004230 Entry: 0x18CFF StackSize: 0x1000 StackMem: 0x20004EB8 SP: 0x2003FD80
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=EP_AGORA
-- MbedOS Error Info --

[image]

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ x] Bug

farrenv, 2019-09-18:
@loverdeg-ep @trowbridgec

trowbridgec, 2019-09-18:
@maclobdell @ARMmbed/team-embeddedplanet

0xc0170, 2019-09-19:
I am missing how to reproduce - why it is failing ?

Looks like the pin is in invalid state, is our gpio_init for this target not initializing pin properly? What's the stack trace?

farrenv, 2019-10-04:
@trowbridgec discovered a #define that limits any nordic target from using more than one Interruptin initialization with SDK 15.0 #11622

closed 2019-10-04 by farrenv