ARMmbed/mbed-os #11519: Can not compile STM32 NUCLEO-F303K8 on mbedOS5

tama14142356, 2019-09-19:
### Description

I am writing some code to toggle the LEDs of a NUCLEO-F303K8 when it receives a message. However, when I try to build, I get the following error. (I am building for mbedOS5)

```
$ mbed compile -m NUCLEO_F303K8 -t GCC_ARM
[mbed] Working path "C:\mbed\vehicle_data_collection\mbed" (library)
[mbed] Program path "C:\mbed\vehicle_data_collection\mbed"
[Warning] @,: Compiler version mismatch: Have 4.9.3; expected version >= 6.0.0 and < 7.0.0
Building project mbed (NUCLEO_F303K8, GCC_ARM)
Scan: mbed

Could not compile for NUCLEO_F303K8: Target does not support mbed OS 5
[mbed] ERROR: "C:\Python27\python.exe" returned error.
       Code: 1
       Path: "C:\mbed\vehicle_data_collection\mbed"
       Command: "C:\Python27\python.exe -u C:\mbed\vehicle_data_collection\mbed\mbed-os\tools\make.py -t GCC_ARM -m NUCLEO_F303K8 --source . --build .\BUILD\NUCLEO_F303K8\GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

For reference, the code I wrote is copied below:

```
#include "mbed.h"
DigitalOut led2(LED2);
// CAN can2(p30, p29);
CAN can2(PA_11, PA_12);
int main() {
   printf("main()\n");
   can2.frequency(500 * 1000);
   CANMessage msg;
   while(1) {
       static int cnt;
       // printf("loop() %d\n", ++cnt);
       while (can2.read(msg)) {
           printf("Message received: id 0x%X, dlc %d [", msg.id, msg.len );
           for (int i = 0; i < msg.len; i++ ) {
               printf( "%02X, ", msg.data[i]);
           }
           printf( "]\n" );
           led2 = !led2;
       }
       wait(0.1);
   }
```

I want to know if NUCLEO_F303K8 is not really supported by mbed OS5
If it is supported, could you tell me how to build it or point me to some resources to learn how?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

Tharazi97, 2019-09-19:
https://os.mbed.com/platforms/ST-Nucleo-F303K8/
It is supported only by Mbed OS 2

Tharazi97, 2019-09-19:
This is a handbook for Mbed OS 2 for CAN:
https://os.mbed.com/handbook/CAN

jeromecoutant, 2019-09-19:
Hi

See ex with https://github.com/ARMmbed/mbed-os-example-blinky-baremetal

Adding
"requires": ["bare-metal"]
in your mbed_app.json file allows you to compile.

PS: I notice your GCC version is quite old...

0xc0170, 2019-09-19:
@tama14142356 Does that answer your question? If yes, please close the issue.

closed 2019-09-20 by tama14142356