ARMmbed/mbed-os #11530: Gaps in QSPIFBlockDevice SFDP Implementation

kyle-cypress, 2019-09-19:
### Description
In several cases QSPIFBlockDevice does not conform to the SFDP standard [JESD216](https://www.jedec.org/system/files/docs/JESD216D-01.pdf).  These include:

 - Status register reading and writing
 - Soft resetting
 - Enabling of 4-byte addressing
 - Enabling of quad output mode
 - Use of the legacy erase instruction

In these cases QSPIFBlockDevice often assumes that certain instructions or methods are universal across flash devices when they are not.  This can cause issues in devices that do not match these assumptions.  We will be submitting a pull request shortly to resolve many of these issues and better align QSPIFBlockDevice with JESD216.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-09-20:
cc @ARMmbed/mbed-os-storage

adbridge, 2019-11-21:
@kyle-cypress can this be closed now ?

kyle-cypress, 2019-11-21:
Yes, the PR #11531 that fixes this has been merged. Thanks.

closed 2019-11-21 by 0xc0170