ARMmbed/mbed-os #11533: What means value of function can.rderror and can.tderror

tama14142356, 2019-09-20:
### Description

my environment is below:

toolchain : 
```
name : arm-none-abi-gcc 
gcc version: 6.3.1
```
tools:  
```
name : mbed-cli
mbed-cli version:1.8.3
```
SHA of mbedOS
```
e3e9aef (HEAD -> master, origin/master, origin/HEAD) Merge pull request #3 from ARMmbed/mbed-os-5.12
```

I wrote some code for the NUCLEO_F303K8 to receive CAN data from an LPC1768.
I'm expecting that the data continuously gets sent, but for some reason it seems to only get sent five times, and the data only gets received once.

receive:
```
main()
loop() 1
setfilter : 1
# of rd error : 0
# of rd error : 0
Message received: id 0x17C, dlc 1 [01, ]
loop() 2
setfilter : 1
# of rd error : 255
# of rd error : 255
Message received: id 0x17C, dlc 1 [01, ]
loop() 3
setfilter : 1
# of rd error : 255
# of rd error : 255
Message received: id 0x17C, dlc 1 [01, ]
loop() 4
setfilter : 1
# of rd error : 255
# of rd error : 255
Message received: id 0x17C, dlc 1 [01, ]
loop() 5
setfilter : 1
# of rd error : 255
# of rd error : 255
Message received: id 0x17C, dlc 1 [01, ]
```

send:
```
main()
send()
wloop()
Message sent: 1
loop()
Message sent counter : 1
# of tderr : 128
send()
wloop()
Message sent: 2
loop()
Message sent counter : 2
# of tderr : 128
send()
wloop()
Message sent: 3
loop()
Message sent counter : 3
# of tderr : 134
send()
wloop()
Message sent: 4
loop()
Message sent counter : 4
# of tderr : 134
send()
wloop()
Message sent: 5
loop()
Message sent counter : 5
# of tderr : 134
send()
loop()
Message sent counter : 5
# of tderr : 134
```

For reference, the code which I wrote is copied below:
receive:
```
/* mbed Microcontroller Library
 * Copyright (c) 2019 ARM Limited
 * SPDX-License-Identifier: Apache-2.0
 */
#include "mbed.h"

DigitalOut led2(LED2);
CAN fcan(PA_11, PA_12);
CANMessage buf;
Serial pc(USBTX, USBRX);

int setFilter(unsigned int id)
{
    int ret = fcan.filter(id, 0);
    return ret;
}
void debugShow()
{
    pc.printf("# of rd error : %d\n", fcan.rderror());
    pc.printf("Message received: id 0x%X, dlc %d [", buf.id, buf.len);
    for (int i = 0; i < buf.len; i++)
    {
        pc.printf("%02X, ", buf.data[i]);
    }
    pc.printf("]\n");
}

void receiveCallback()
{
    CANMessage msg;
    if (fcan.read(msg))
    {
        buf.id = msg.id;
        buf.len = msg.len;
        for (int i = 0; i < msg.len; ++i)
        {
            buf.data[i] = msg.data[i];
        }
        led2 = !led2;
    }
}

int main()
{
    pc.printf("main()\n");

    fcan.frequency(500 * 1000);
    int ret = setFilter(0x091);
    fcan.attach(receiveCallback);

    while (1)
    {
        static int cnt;
        pc.printf("loop() %d\n", ++cnt);
        pc.printf("setfilter : %d\n", ret);
        pc.printf("# of rd error : %d\n", fcan.rderror());
        debugShow();
        wait(1);
    }
}

```

send:
```
#include "mbed.h"
//sender
Ticker ticker;
DigitalOut led1(LED1);
DigitalOut led2(LED2);
// CAN can1(p9, p10);
CAN can2(p30, p29);
char counter = 0;
 
void send() {
    printf("send()\n");
    if(can2.write(CANMessage(0x17C, &counter, 1))) {
        printf("wloop()\n");
        counter++;
        printf("Message sent: %d\n", counter);
    } 
    led1 = !led1;
}
 
int main() {
    printf("main()\n");
    // ticker.attach(&send, 1);
    CANMessage msg;
    while(1) {
        send();
        printf("loop()\n");
        printf("Message sent counter : %d\n", counter);
        printf("# of tderr : %d\n", can2.tderror());
        wait(0.1);
    }
}

```

So, I used can.rderror and can.tderror to debug the error while sending and receiving. However, I didn't understand the values I was getting. (You can see # of tderr and # of of rd error in my output.)
Could anyone explain what the values given by can.rderror and can.tderror mean, or point me to some resource to learn about them?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

Tharazi97, 2019-09-20:
I might be wrong, but if you are using CAN interrupts, doesn't you have to attach them for each time you want to receive a message? I mean shouldn't the ``receiveCallback()`` use ``fcan.attach(receiveCallback);``?

jeromecoutant, 2019-10-02:
Hi
I am not a CAN expert at all...

mbed API: https://github.com/ARMmbed/mbed-os/blob/31114ba3436014619d157beadde7a5294f10f54b/drivers/CAN.h#L252

In ST side, rderror returns the "Receive error counter" from CAN error status register.

Looking at reference manual:
The implementing part of the fault confinement mechanism of the CAN protocol. In case of
an error during reception, this counter is incremented by 1 or by 8 depending on the error
condition as defined by the CAN standard. After every successful reception the counter is
decremented by 1 or reset to 120 if its value was higher than 128. When the counter value
exceeds 127, the CAN controller enters the error passive state.

Not clear for me...

tama14142356, 2019-10-05:
Thank you for answering my question.

I could manage to understand.

closed 2019-10-05 by tama14142356