ARMmbed/mbed-os #11537: ESP8266: Pelion connect fails with scanf(...)

ccli8, 2019-09-20:
### Description

#### Environment

**mbed-os-example-pelion**
656809a Merge pull request #14 from ARMmbed/remove-patches

**mbed-clout-client**
3.4.0

**mbed-os**
83fca603f0 Merge pull request #11454 from Tharazi97/LSI_VALUE_STM

**Target**
NUMAKER_IOT_M263A

#### Related PR

#11459 

During my pelion test, I find something wrong with esp8266 driver. For my trace, I add `printf(...)` calls in:

**mbed-os/components/wifi/esp8266-driver/ESP8266/ESP8266.cpp** > **ESP8266::_oob_packet_hdlr()**:

<pre>

if (_tcp_passive && _sock_i[id].open == true && _sock_i[id].proto == NSAPI_TCP) {
        if (_parser.scanf("%d\n", &amount)) {
            _sock_i[id].tcp_data_avbl = amount;

            <b>
            //TESTTEST
            printf("MYCHECK: TCP: amount: %d\r\n", amount);
            </b>

            // notify data is available
            if (_callback) {
                _callback();
            }
        }
        return;
    } else if (!_parser.scanf("%d:", &amount)) {
        return;
    }

    <b>
    //TESTTEST
    printf("\r\nMYCHECK: UDP: amount: %d\r\n\r\n", amount);
    </b>
</pre>

Pelion connect fails:

<pre>

Mbed Bootloader
No Update image
[DBG ] Active firmware up-to-date
booting...

Initialize KVStore
Connect to network
Network connected with IP 192.168.8.107

Start developer flow
Create resources
Register Pelion Device Management Client

MYCHECK: UDP: amount: 109

MYCHECK: TCP: amount: 7
MYCHECK: TCP: amount: 9
client_error(6) -> Client in reconnection mode NetworkError
MYCHECK: TCP: amount: 7
MYCHECK: TCP: amount: 9
client_error(6) -> Client in reconnection mode NetworkError

MYCHECK: UDP: amount: 109

MYCHECK: TCP: amount: 7
MYCHECK: TCP: amount: 9
client_error(6) -> Client in reconnection mode NetworkError
MYCHECK: TCP: amount: 7
MYCHECK: TCP: amount: 8

</pre>

If I change one `scanf()` back to `recv()`:

**mbed-os/components/wifi/esp8266-driver/ESP8266/ESP8266.cpp** > **ESP8266::_oob_packet_hdlr()**:

<pre>

if (_tcp_passive && _sock_i[id].open == true && _sock_i[id].proto == NSAPI_TCP) {
        if (_parser.<b>recv</b>("%d\n", &amount)) {
            _sock_i[id].tcp_data_avbl = amount;

</pre>

Pelion connect succeeds:

<pre>

Mbed Bootloader
No Update image
[DBG ] Active firmware up-to-date
booting...

Initialize KVStore
Connect to network
Network connected with IP 192.168.8.107

Start developer flow
Create resources
Register Pelion Device Management Client

MYCHECK: UDP: amount: 109

MYCHECK: TCP: amount: 755
MYCHECK: TCP: amount: 186
MYCHECK: TCP: amount: 43
MYCHECK: TCP: amount: 811
MYCHECK: TCP: amount: 1046
MYCHECK: TCP: amount: 59
MYCHECK: TCP: amount: 60
MYCHECK: TCP: amount: 1065
MYCHECK: TCP: amount: 553
MYCHECK: TCP: amount: 549
MYCHECK: TCP: amount: 482
MYCHECK: TCP: amount: 42
MYCHECK: TCP: amount: 32
MYCHECK: TCP: amount: 25
MYCHECK: TCP: amount: 48

MYCHECK: UDP: amount: 106

MYCHECK: TCP: amount: 96
MYCHECK: TCP: amount: 744
MYCHECK: TCP: amount: 925
MYCHECK: TCP: amount: 43
MYCHECK: TCP: amount: 32
MYCHECK: TCP: amount: 32
MYCHECK: TCP: amount: 106
MYCHECK: TCP: amount: 141
Client registered.
Account ID: 015e5bd6c70c02420a01140500000000
Endpoint name: 016d4d8a204b000000000001001002de
Device ID: 016d4d8a204b000000000001001002de

</pre>

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-09-20:
@dmaziec1 Please review

cc @ARMmbed/mbed-os-ipcore

michalpasztamobica, 2019-09-20:
@ccli8 , thanks for reporting! 
Somehow, we haven't seen this with K64F tests and CI hasn't noticed anything before the merge, but we will look into this and try to figure out a fix.

ccli8, 2019-10-01:
Fixed via #11541

closed 2019-10-01 by ccli8