ARMmbed/mbed-os #11543: STM32 clock security system for LSE

maxgerhardt, 2019-09-22:
### Description

- target: NUCLEO_L476RG 
- tools: mbed-cli 
- release: 5.13.3

Currently, the implementation of the 'Clock Security System' (for both LSE/LSI and HSE/HSI) for the STM32 MCUs is lacking in mbed-os. In fact the LSE/LSI CSS is even disabled in the MSI clock setup code and never re-enabled. 

https://github.com/ARMmbed/mbed-os/blob/daafb950d96e2e72027f72f1dbdfdd6333cf7fbc/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L476xG/TARGET_NUCLEO_L476RG/system_clock.c#L279-L280

For hardening devices, the clock security system is important. However, the current mbed-os implementation makes it hard to implement CSS callbacks properly. As the [reference manual](https://www.st.com/content/ccc/resource/technical/document/reference_manual/02/35/09/0c/4f/f7/40/03/DM00083560.pdf/files/DM00083560.pdf/jcr:content/translations/en.DM00083560.pdf) for STM32L4x6 device says in chapter 6.2.11: 

> A Clock Security System on LSE can be activated by software writing the LSECSSON bit in
the Backup domain control register (RCC_BDCR). This bit can be disabled only by a
hardware reset or RTC software reset, or after a failure detection on LSE. LSECSSON must
be written after LSE and LSI are enabled (LSEON and LSION enabled) and ready (LSERDY
and LSIRDY set by hardware), and after the RTC clock has been selected by RTCSEL.
> [..] 
> The software MUST then disable the LSECSSON bit, stop the defective 32 kHz oscillator
(disabling LSEON), and change the RTC clock source (no clock or LSI or HSE, with
RTCSEL), or take any required action to secure the application.

So even if I call `HAL_RCCEx_EnableLSECSS()` and implement the `_WEAK` callback [`HAL_RCCEx_LSECSS_Callback()`](https://github.com/ARMmbed/mbed-os/blob/daafb950d96e2e72027f72f1dbdfdd6333cf7fbc/targets/TARGET_STM/TARGET_STM32L4/device/stm32l4xx_hal_rcc_ex.c#L2390-L2395) in my app, I have to write my own RTC clock init routine because `rtc_init()` has hard-coded `#ifdefs` for whether the LSE is available on the system. 

https://github.com/ARMmbed/mbed-os/blob/daafb950d96e2e72027f72f1dbdfdd6333cf7fbc/targets/TARGET_STM/rtc_api.c#L61-L86

mbed-os is hardcoded to assume either LSE or LSI will be there forever. It would be nice if mbed-os integrate's the CSS system and reinitializes peripherals (here: RTC and LP ticker, [which you hardcoded as well](https://github.com/ARMmbed/mbed-os/blob/daafb950d96e2e72027f72f1dbdfdd6333cf7fbc/targets/TARGET_STM/lp_ticker.c#L105-L130)) on clock failure to make the firmwares more resilient by default.

Similar for high-speed clock sources. Though `SetSysClock()` is called on bootup and on every wakeup from deepsleep, but what if during the "run" section of a firmware, we suddenly lose the external high-speed clock source (HSE)?

### Issue request type

    [] Question
    [X] Enhancement
    [ ] Bug

jeromecoutant, 2019-10-01:
Hi
I agree we should enable LSE CSS should be enabled.
But I think we could propose to change RTC and LPTICKER with LSI, this will increase code size ?

maxgerhardt, 2019-10-03:
>  But I think, if we propose to change automatically RTC and LPTICKER configuration with LSI, this will increase code size ? 

Definitely, since code excluded by the compile-time macro `MBED_CONF_TARGET_LSE_AVAILABLE` is now also enabled. I think functions like `rtc_init()` should get a parameter whether initialization is to be done with LSE or LSI, then all the `#ifdef ..LSE_AVAILABLE` become `if(lse_available)` via a parameter. Same for LPTICKER and the  hardcoded frequency values noted above. However the code size increase shouldn't really be that much, since only a new branch exists with some extra code / parameter changes.

0xc0170, 2019-10-22:
> Note: #7787 correction to update

@jeromecoutant What is left for this one, all resolved?

jeromecoutant, 2019-10-22:
> What is left for this one, all resolved?

For this one, I agree we need to enable clock security system.
What is not decided yet, is the behavior the default mbed code should have in case of clock issue...

@maxgerhardt maybe you could propose in a PR what you are expecting?

@LMESTM

closed 2019-10-31 by ciarmcom