ARMmbed/mbed-os #11552: Cellular - AtHandler fails to read sim

jensoleg, 2019-09-23:
### Description

Customer target with U-BLOX R410M modem. mbed version is 5.13.4

I am trying to get the Cellular network up but it fails when checking the sim state. This is a snippet from the log
```
[DBG ][CELL]: Device ready => SIM pin
[DBG ][CELL]: callback: 4106, err: 0, data: 4097
[INFO][CELL]: Setup SIM (timeout 1000 ms)
[DBG ][CELL]: AT flush
[INFO][CELL]: AT TX ( 8): AT+CPIN?
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX (20): <cr>CPIN: READY<cr><ln><cr><ln>OK<cr><ln>
[WARN][CELL]: SIM not readable. Length is -1
[WARN][CELL]: SIM state unknown
[DBG ][CELL]: callback: 4097, err: 0, data: 3
```
 
I tested on the U-blox CO30 development board and it works just fine 
```
[00000131ms][DBG ][CELL]: Device ready => SIM pin
[00000136ms][DBG ][CELL]: callback: 4106, err: 0, data: 4097
[00000143ms][INFO][CELL]: Setup SIM (timeout 1000 ms)
[00000149ms][DBG ][CELL]: AT flush
[00000153ms][INFO][CELL]: AT TX ( 8): AT+CPIN?
[00000159ms][INFO][CELL]: AT TX ( 1): <cr>
[00000164ms][INFO][CELL]: AT RX (22): <cr><ln>+CPIN: READY<cr><ln><cr><ln>OK<cr><ln>
[00000172ms][DBG ][CELL]: SIM read: READY
```
I notice that the response on the custom board swallows the `<ln>+` in the CPIN response. If I force the network connection process to continue I see that these char is also missing in other responses -  the f.ex. CSQ.

I also tried to request CPIN with the AtCmdParser on the custom board and the response is OK:

```
AT> AT+CPIN?
AT? +CPIN: %*15[^
]
%n
AT< 
AT< AT+CPIN?
AT< 
AT= +CPIN: READY

AT? OK%n
AT= OK 
```

### Issue request type
    [x] Question
    [ ] Enhancement
    [ ] Bug

kivaisan, 2019-10-09:
@jensoleg Does it happen with AT-traces disabled? You can also try to increase UART RX buffer with setting `"drivers.uart-serial-rxbuf-size": 1024,` in your `mbed_app.json`

jensoleg, 2019-10-14:
HI @kivaisan - I tried both your suggestions but without luck.

Will try to revert the code to use the UBLOX libraries instead of the mbed cellular feature. 

```
[00003145ms][INFO][CELL]: New CellularContext  (0x20004bc8)
Establishing connection
[00003155ms][INFO][CELL]: CellularContext connect
[00003164ms][DBG ][CELL]: callback: 4106, err: 0, data: 4096
[00003172ms][INFO][CELL]: Start connecting (timeout 1000 ms)
[00004180ms][DBG ][CELL]: AT error -3012
[00004187ms][DBG ][CELL]: AT error -3012
[00004193ms][DBG ][CELL]: Init => Power
[00004199ms][DBG ][CELL]: AT OoB readable 1, len 0
[00004227ms][DBG ][CELL]: AT OoB done
[00004233ms][DBG ][CELL]: callback: 4106, err: 0, data: 4096
[00004240ms][INFO][CELL]: Modem power ON (timeout 1000 ms)
[00004248ms][DBG ][CELL]: Power => Device ready
[00004255ms][DBG ][CELL]: callback: 4106, err: 0, data: 4096
[00004262ms][DBG ][CELL]: Device was not ready, calling soft_power_on()
[00004271ms][DBG ][CELL]: AT flush
[00004290ms][INFO][CELL]: Modem ready
[00004295ms][DBG ][CELL]: callback: 4096, err: 0, data: 4096
[00004303ms][DBG ][CELL]: Device ready => SIM pin
[00004310ms][DBG ][CELL]: callback: 4106, err: 0, data: 4097
[00004318ms][INFO][CELL]: Setup SIM (timeout 1000 ms)
[00004325ms][DBG ][CELL]: AT flush
[00005326ms][DBG ][CELL]: AT error -3012
[00005332ms][WARN][CELL]: SIM not readable.
[00005338ms][WARN][CELL]: SIM state unknown
[00005343ms][INFO][CELL]: Waiting for SIM (err while reading)...
[00005351ms][DBG ][CELL]: SIM pin: retry 0/10
[00005358ms][DBG ][CELL]: callback: CellularStateRetryEvent, err: 0, data: 4097, retrycount: 0
[00005368ms][INFO][CELL]: Continue after 1 seconds
[00006375ms][DBG ][CELL]: callback: 4106, err: 0, data: 4097
[00006383ms][INFO][CELL]: Setup SIM (timeout 1000 ms)
[00006390ms][DBG ][CELL]: AT flush
[00006402ms][WARN][CELL]: SIM not readable.
[00006408ms][WARN][CELL]: SIM state unknown
[00006414ms][DBG ][CELL]: callback: 4097, err: 0, data: 3
[00006421ms][DBG ][CELL]: SIM pin: retry 1/10
[00006428ms][DBG ][CELL]: callback: CellularStateRetryEvent, err: 0, data: 4097, retrycount: 1
[00006438ms][INFO][CELL]: Continue after 2 seconds
[00008445ms][DBG ][CELL]: callback: 4106, err: 0, data: 4097
[00008453ms][INFO][CELL]: Setup SIM (timeout 1000 ms)
[00008460ms][DBG ][CELL]: AT flush
[00008471ms][WARN][CELL]: SIM not readable.
[00008477ms][WARN][CELL]: SIM state unknown
````

closed 2019-10-25 by ciarmcom