ARMmbed/mbed-os #11555: Cannot use queue

tama14142356, 2019-09-24:
### Description

my environment is below:

toolchain : 
```
name : arm-none-abi-gcc 
gcc version: 6.3.1
```
tools:  
```
name : mbed-cli
mbed-cli version:1.8.3
```
SHA of mbedOS
```
e3e9aef (HEAD -> master, origin/master, origin/HEAD) Merge pull request #3 from ARMmbed/mbed-os-5.12
```

I am writing some code to store the data using FI-FO data structure.  
However, when I try to build, I get the following error. 

```
$ mbed compile
[mbed] Working path "C:\mbed\vehicle_data_collection\stm_receive" (library)
[mbed] Program path "C:\mbed\vehicle_data_collection\stm_receive"
Building project stm_receive (NUCLEO_F303K8, GCC_ARM)
Scan: stm_receive
Compile [100.0%]: main.cpp
[Error] main.cpp@50,5: 'Queue' was not declared in this scope
[Error] main.cpp@50,11: expected primary-expression before 'char'
[Error] main.cpp@57,9: 'queue' was not declared in this scope
[Error] main.cpp@58,9: 'osEvent' was not declared in this scope
[Error] main.cpp@59,13: 'evt' was not declared in this scope
[Error] main.cpp@59,20: 'osEventMessage' was not declared in this scope
[ERROR] .\main.cpp: In function 'int main()':
.\main.cpp:50:5: error: 'Queue' was not declared in this scope
     Queue<char, 10> queue;
     ^~~~~
.\main.cpp:50:11: error: expected primary-expression before 'char'
     Queue<char, 10> queue;
           ^~~~
.\main.cpp:57:9: error: 'queue' was not declared in this scope
         queue.put('1');
         ^~~~~
.\main.cpp:58:9: error: 'osEvent' was not declared in this scope
         osEvent evt = queue.get();
         ^~~~~~~
.\main.cpp:59:13: error: 'evt' was not declared in this scope
         if (evt == osEventMessage)
             ^~~
.\main.cpp:59:20: error: 'osEventMessage' was not declared in this scope
         if (evt == osEventMessage)
                    ^~~~~~~~~~~~~~

[mbed] ERROR: "C:\Python27\python.exe" returned error.
       Code: 1
       Path: "C:\mbed\vehicle_data_collection\stm_receive"
       Command: "C:\Python27\python.exe -u C:\mbed\vehicle_data_collection\stm_receive\mbed-os\tools\make.py -t GCC_ARM -m NUCLEO_F303K8 --source . --build .\BUILD\NUCLEO_F303K8\GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

For reference, the code I wrote and the URL I saw to write the code are copied below:

URL:
[Queue - APIs | Mbed OS 5 Documentation](https://os.mbed.com/docs/mbed-os/v5.13/apis/queue.html)
[Mbed OS Reference | Queue< T, queue_sz > Class Template ..](https://os.mbed.com/docs/mbed-os/v5.13/mbed-os-api-doxy/classrtos_1_1_queue.html)
code :
```
/* mbed Microcontroller Library
 * Copyright (c) 2019 ARM Limited
 * SPDX-License-Identifier: Apache-2.0
 */

#include "mbed.h"

DigitalOut led2(LED2);
CAN fcan(PA_11, PA_12);
CANMessage buf;
Serial pc(USBTX, USBRX); 
int setFilter(unsigned int id)
{
    int ret = fcan.filter(id, 0);
    return ret;
}
void debugShow()
{
    pc.printf("# of rd error : %d\n", fcan.rderror());
    pc.printf("Message received: id 0x%X, dlc %d [", buf.id, buf.len);
    for (int i = 0; i < buf.len; i++)
    {
        pc.printf("%02X, ", buf.data[i]);
    }
    pc.printf("]\n");
}

void receiveCallback()
{
    CANMessage msg;
    if (fcan.read(msg))
    {
        buf.id = msg.id;
        buf.len = msg.len;
        for (int i = 0; i < msg.len; ++i)
        {
            buf.data[i] = msg.data[i];
        }
        led2 = !led2;
    }
}

int main()
{
    pc.printf("main()\n");

    fcan.frequency(500 * 1000);
    int ret = setFilter(0x091);
    fcan.attach(receiveCallback, CAN::RxIrq);
    Queue<char, 10> queue;
    while (1)
    {
        static int cnt;
        pc.printf("loop() %d\n", ++cnt);
        pc.printf("setfilter : %d\n", ret);
        pc.printf("# of rd error : %d\n", fcan.rderror());
        queue.put('1');
        osEvent evt = queue.get();
        if (evt == osEventMessage)
        {
            char *tmp = (char *)evt.value.p;
            pc.printf("queue: %c\n", tmp[0]);
        }
        else
        {
            pc.printf("error\n");
        }
        debugShow();
        wait(0.1);
    }
}

```

I want to know if Queue is able to be used by mbedOS5.
If it can be used, could you tell me how to use it or point me to some resources to learn how?
If it cannot be used it, could you tell me the other data structure which can be used by mbedOS5 or point me to some resources to learn how? 

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

Tharazi97, 2019-09-24:
NUCLEO_F303K8 does not support Mbed OS 5.

https://os.mbed.com/platforms/ST-Nucleo-F303K8/

This is a handbook of queue for Mbed OS 2: https://os.mbed.com/handbook/RTOS

I'm flabbergasted that you haven't receive this message while compiling ``Could not compile for NUCLEO_F303K8: Target does not support mbed OS 5``. If you still want to continue using Mbed os 5, try inluding ``Queue.h`` maybe it will help.

0xc0170, 2019-09-24:
> I'm flabbergasted that you haven't receive this message while compiling Could not compile for NUCLEO_F303K8: Target does not support mbed OS 5. If you still want to continue using Mbed os 5, try inluding Queue.h maybe it will help.

It could be that the app is using Mbed 2 and thus can't find the requested files, however the first comment above references Mbed OS...

closed 2019-10-17 by tama14142356