ARMmbed/mbed-os #11565: Profiles contain RTX symbol (bare metal error)

0xc0170, 2019-09-25:
### Description

This PR https://github.com/ARMmbed/mbed-os/pull/8887 added debug awareness - linker flag addition `keep` a section that is RTX specific. This introduced a warning to the baremetal build and mbed 2 - there is no `os_cb_sections`. 

Reading this comment: https://github.com/ARMmbed/mbed-os/pull/8887/files#r243106755 , is this still needed? 

One interesting fact about this warning: users think this warning causes build failing in the online compiler. There are so many reports about this, see https://os.mbed.com/questions/85360/Error-Warning-L6320W-ignoring-keep-comma/#answer16979 , https://os.mbed.com/questions/86252/Cant-Compile-Error-Warning-L6320W-Ignori/ and others. The real error there is not always shown, leads a user to believe this warning is actually an error and causing their apps not to build.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-09-25:
cc @ARMmbed/mbed-os-core

evedon, 2019-09-26:
We should be able to work on this shortly as we are looking at changing linker options for bare metal.
cc @rajkan01

PR #11699 (2019-10-18): Fix for missing os_cb_section
 -Added the os_cb_section stub to remove the warnings

### Description
Procedure to reproduce the issue:
1.Clone the blinky bare metal from https://github.com/ARMmbed/mbed-os-example-blinky-baremetal.git 
2. Compile the blinky bare metal with "mbed compile -t ARMC6 -m target --profile release"
3.End of the compilation you will notice "[Warning] @0,0: L6320W: Ignoring --keep command. Cannot find argument 'os_cb_sections'." warning.

Reason for the issue:
1. In release profile, ARMC6 compiler have extra optional argument --keep=os_cb_sections but os_cb_section defin [... cut]
closed 2019-10-18 by 0xc0170