ARMmbed/mbed-os #11570: Greentea tests fail to compile in baremetal mode in Mbed OS 5.14

MarceloSalazar, 2019-09-26:
### Description

The baremetal mode was introduced in [Mbed OS 5.12](https://github.com/ARMmbed/mbed-os/releases/tag/mbed-os-5.12.0) and we need to be able to run tests to confirm it works well across a number of platforms.

From the [baremetal](https://github.com/ARMmbed/mbed-os/blob/master/platform/bare_metal/mbed_lib.json) profile, there are two main items that are included: ["platform", "drivers"]

However, there are compile errors when trying to build simple greentea tests for drivers/platform on the [Mbed OS baremetal](https://os.mbed.com/docs/mbed-os/v5.14/reference/mbed-os-bare-metal.html) profile.

### Steps to reproduce

```
mbed import https://github.com/ARMmbed/mbed-os-example-blinky-baremetal#mbed-os-5.14
cd mbed-os-example-blinky-baremetal
move main.cpp main.old
mbed test -t ARM -m NUCLEO_F429ZI -n *echo* --compile

...

Build failures:
  * NUCLEO_F429ZI::ARMC6::MBED-OS-TESTS-MBED_DRIVERS-ECHO
        Building project echo (NUCLEO_F429ZI, ARMC6)
        Scan: ARM
        Scan: echo
        Compile [100.0%]: main.cpp
        [Fatal Error] main.cpp@20,10: 'greentea-client/test_env.h' file not found
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-09-26:
Thanks for reporting, we really need this to be caught in CI tests
cc @ARMmbed/mbed-os-test

jeromecoutant, 2019-09-26:
Thx @MarceloSalazar 
From my side, I would like to be able to execute tests on non-OS5 targets (MBED2 only ones)

adbridge, 2019-09-26:
Coudl this be related to the header files changes introduced as part of 5.14?

MarceloSalazar, 2019-09-26:
I can reproduce the same problem in 5.12

adbridge, 2019-09-26:
> 
> 
> I can reproduce the same problem in 5.12

So it has always been broken :(

0xc0170, 2019-09-26:
Wasn't this the case since baremetal was introduced ?

https://os.mbed.com/docs/mbed-os/v5.13/reference/mbed-os-bare-metal.html - this does not state if any of frameworks are available (greentea and others). It should be available but why is not? @ARMmbed/mbed-os-tools Please review

cc @ARMmbed/mbed-os-core - you've been working on fixing some of the baremetal fixes recently

evedon, 2019-09-26:
cc @jamesbeyond mentioned that there is some work in progress to add bare metal support in greentea

pdemil, 2019-10-10:
Out of the box experience: 
https://os.mbed.com/docs/mbed-os/v5.14/quick-start/compiling-the-code.html 
you can read:
"To build with the Mbed OS bare metal profile, add "requires": ["bare-metal"] to the mbed_app.json file." 
Adding this does not work out of the box because of following error:
[Error] main.cpp@27,9: 'ThisThread' has not been declared

There is a bare metal blinky example here: 
mbed import https://github.com/ARMmbed/mbed-os-example-blinky-baremetal
but this example is still using wait_ms (deprecated since 5.14).

evedon, 2019-10-18:
> Out of the box experience:
> https://os.mbed.com/docs/mbed-os/v5.14/quick-start/compiling-the-code.html
> you can read:
> "To build with the Mbed OS bare metal profile, add "requires": ["bare-metal"] to [...]

Thanks for pointing this out. We should update the documentation to point at the bare metal blinky example: https://github.com/ARMmbed/mbed-os-example-blinky-baremetal
and replace the deprecated `wait_ms` call by `wait_us`

jamesbeyond, 2019-10-18:
In term of testing, Greentea is not fully support Baremetal mode yet, we are in the progress of enable the that function. we are looking to finish that before next mbed-os minor release. once it is done, A documentation about how to run Greentea bare-metal test will be published, I'll attached the link here

evedon, 2019-10-18:
> Thanks for pointing this out. We should update the documentation to point at the bare metal blinky example: https://github.com/ARMmbed/mbed-os-example-blinky-baremetal
> and replace the deprecated [...]

@AnotherButler Can you update the doc please?

AnotherButler, 2019-10-18:
We point to the bare metal Blinky in the bare metal section: https://os.mbed.com/docs/mbed-os/v5.14/reference/bare-metal-example-application.html

Or are you saying the standard quick start (https://os.mbed.com/docs/mbed-os/v5.14/quick-start/index.html) should point to the bare metal example?

AnotherButler, 2019-10-30:
@evedon Could you please clarify the above?

evedon, 2019-10-31:
The standard quick start should point to the bare metal example and the instructions on how to build the bare metal profile found here https://os.mbed.com/docs/mbed-os/v5.14/quick-start/compiling-the-code.html should move to the bare metal example section.

evedon, 2020-01-07:
Baremetal support was added to greentea in 5.15 so the original issue can be closed.

To run greentea tests in baremetal mode, the command to use is:
```
mbed test -t ARM -m NUCLEO_F429ZI --app-config TESTS/configs/baremetal.json
```

closed 2020-01-07 by evedon