ARMmbed/mbed-os #11574: Gap::addDeviceToPeriodicAdvertiserList always return BLE_ERROR_INVALID_PARAM

anton-veretenenko, 2019-09-26:
### Description

Trying to sync to several peers who use periodic advertising. Sync to single peer works.
Then I try to use Gap::addDeviceToPeriodicAdvertiserList and Gap::createSync to the list.
But Gap::addDeviceToPeriodicAdvertiserList always return BLE_ERROR_INVALID_PARAM after checking peerAddressType even if I provide proper peer address type.
I took a look in the code and found this:

`
    if (peerAddressType != PeerAddressType_t::PUBLIC ||
        peerAddressType != PeerAddressType_t::RANDOM
        ) {
        return BLE_ERROR_INVALID_PARAM;
    }
`
at /features/FEATURE_BLE/source/generic/GenericGap.tpp
This will always return BLE_ERROR_INVALID_PARAM even if you provide proper PUBLIC or RANDOM peer address type.

After fixing this if statement, sync to the list does not work, after createSync called there are no events at all, no timeouts, nothing.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ x] Bug

0xc0170, 2019-11-06:
cc @ARMmbed/mbed-os-pan

paul-szczepanek-arm, 2020-10-28:
Fixed in https://github.com/ARMmbed/mbed-os/pull/13822 but further issues remain with creating syncs based on the list. This is tracked internally and will be addressed asap.

paul-szczepanek-arm, 2020-12-01:
Turned out to be quite hard to track down, PR is up:
https://github.com/ARMmbed/mbed-os/pull/13988

paul-szczepanek-arm, 2020-12-10:
This is now fixed on master.

closed 2020-12-10 by 0xc0170