ARMmbed/mbed-os #11575: Gap::terminateSync do nothing

anton-veretenenko, 2019-09-26:
### Description

Using your example at https://github.com/ARMmbed/mbed-os-example-ble/blob/master/BLE_PeriodicAdvertising
If I add a call to Gap::terminateSync after some time after sync has been established to stop receiving periodic advertising reports then nothing happens. Sync does not terminate, I still receive periodic advertising reports from the peer and can not sync to other peer.

Target is NRF52840 custom board. Mbed 5.13.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-09-30:
cc @ARMmbed/mbed-os-pan

paul-szczepanek-arm, 2021-02-10:
(proposed) fix https://github.com/ARMmbed/mbed-os/pull/14264

paul-szczepanek-arm, 2021-02-12:
OK, this is now merged, I verified it on the example but can you please check it works for you @anton-veretenenko?

closed 2021-02-25 by ciarmcom