ARMmbed/mbed-os #11579: STM32H743ZI Unable to use AXI SRAM mapped at address 0x2400 0000.

kpyush, 2019-09-27:
### Description
branch: mbed-os-5.12
Target: targets/TARGET_STM/TARGET_STM32H7
Toolchain: GCC_ARM
Cli Version: version 1.10.1

**Reproduction Steps:** 
1. Change the RAM address in [linker file ](https://github.com/ARMmbed/mbed-os/blob/73f096399b4cda1f780b140c87afad9446047432/targets/TARGET_STM/TARGET_STM32H7/TARGET_STM32H743xI/device/TOOLCHAIN_GCC_ARM/STM32H743xI.ld#L22) from 0x20000298 to 0x24000298
2. Run [mbed-os-example-blinky](https://github.com/ARMmbed/mbed-os-example-blinky)

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jeromecoutant, 2019-09-27:
Hi
I think it is already done in the master branch ?

@VVESTM

kpyush, 2019-09-27:
Yes its fixed in the master branch but the thing is  I am using 5.12.0 release and I don't want to upgrade the Mbed As of now. Is there any way that I can use the AXI_SRAM in this release only?

0xc0170, 2019-09-30:
> Yes its fixed in the master branch but the thing is I am using 5.12.0 release and I don't want to upgrade the Mbed As of now. Is there any way that I can use the AXI_SRAM in this release only?

You can cherry-pick specific commit or just pull the entire changeset (in this case pull request) to your working version. As this is a target update, should be straightforward.

@jeromecoutant Do we know the specific PR number to post here, we shall then close the issue as resolved.

VVESTM, 2019-09-30:
Hi @kpyush ,
You must use PR #11562.

closed 2019-09-30 by 0xc0170