ARMmbed/mbed-os #11600: STM32H7xx - dereferencing type-punned pointers

JanneKiiskila, 2019-09-30:
### Description

STM32H7xx drivers `stm32h7xx_hal_i2s.c` and `stm32h7xx_hal_spi.c` have multiple compilers warnings on type-punned pointers.

Using current tip of master of mbed-os; 1319843042 (origin/master, origin/HEAD, master) Merge pull request #11590 from morser499/pr/udb-sdio-init

GCC ARM v8

Any compilation will do (since `mbed cli` insists on compiling absolutely everything), I'm using a very short serial test.

```
#include "mbed.h"

int main() {
    printf("test\n");
    NVIC_SystemReset();
}
```
This provides the following warnings:

```
Compile [ 88.8%]: stm32h7xx_hal_i2s.c
[Warning] stm32h7xx_hal_i2s.c@823,9: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
[Warning] stm32h7xx_hal_i2s.c@932,43: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
[Warning] stm32h7xx_hal_i2s.c@1710,5: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
[Warning] stm32h7xx_hal_i2s.c@1766,39: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
Compile [ 93.2%]: stm32h7xx_hal_spi.c
[Warning] stm32h7xx_hal_spi.c@885,13: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
[Warning] stm32h7xx_hal_spi.c@924,13: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
[Warning] stm32h7xx_hal_spi.c@1099,47: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
[Warning] stm32h7xx_hal_spi.c@1139,47: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
[Warning] stm32h7xx_hal_spi.c@1330,13: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
[Warning] stm32h7xx_hal_spi.c@1349,47: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
[Warning] stm32h7xx_hal_spi.c@1388,13: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
[Warning] stm32h7xx_hal_spi.c@1414,47: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
[Warning] stm32h7xx_hal_spi.c@2786,49: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
[Warning] stm32h7xx_hal_spi.c@3434,39: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
[Warning] stm32h7xx_hal_spi.c@3544,5: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
```
    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-11-06:
> Attn: @ARMmbed/team-st-mcd

Was there any internal ticket created for this? How can we fix these warnings?

jeromecoutant, 2019-11-06:
ST_INTERNAL_REF 75246

jeromecoutant, 2020-01-21:
Hi
It should be corrected in
https://github.com/STMicroelectronics/STM32CubeH7

0xc0170, 2020-05-08:
@jeromecoutant Will we get  the update here to close this issue?

jeromecoutant, 2020-05-11:
Hi
Planned after #12801

jeromecoutant, 2020-06-08:
Fixed in #13083

JanneKiiskila, 2020-06-29:
Indeed, compilation is now very clean for the STM files, thanks!

```
...
Compile [ 84.2%]: stm32h7xx_hal.c
Compile [ 84.3%]: Thread.cpp
Compile [ 84.4%]: stm32h7xx_hal_adc.c
Compile [ 84.5%]: ThisThread.cpp
Compile [ 84.6%]: stm32h7xx_hal_adc_ex.c
Compile [ 84.7%]: stm32h7xx_hal_cec.c
Compile [ 84.8%]: stm32h7xx_hal_comp.c
Compile [ 84.9%]: stm32h7xx_hal_cortex.c
Compile [ 85.0%]: stm32h7xx_hal_crc.c
Compile [ 85.1%]: stm32h7xx_hal_crc_ex.c
Compile [ 85.2%]: stm32h7xx_hal_cryp.c
Compile [ 85.3%]: stm32h7xx_hal_cryp_ex.c
Compile [ 85.4%]: stm32h7xx_hal_dac.c
Compile [ 85.5%]: stm32h7xx_hal_dac_ex.c
Compile [ 85.6%]: stm32h7xx_hal_dcmi.c
Compile [ 85.7%]: stm32h7xx_hal_dfsdm_ex.c
Compile [ 85.8%]: stm32h7xx_hal_dfsdm.c
Compile [ 86.0%]: stm32h7xx_hal_dma2d.c
Compile [ 86.1%]: stm32h7xx_hal_dma.c
Compile [ 86.2%]: stm32h7xx_hal_dma_ex.c
Compile [ 86.3%]: stm32h7xx_hal_dts.c
Compile [ 86.4%]: stm32h7xx_hal_dsi.c
Compile [ 86.5%]: stm32h7xx_hal_eth_ex.c
Compile [ 86.6%]: stm32h7xx_hal_exti.c
Compile [ 86.7%]: stm32h7xx_hal_eth.c
Compile [ 86.8%]: stm32h7xx_hal_flash.c
Compile [ 86.9%]: stm32h7xx_hal_gfxmmu.c
Compile [ 87.0%]: stm32h7xx_hal_fdcan.c
Compile [ 87.1%]: stm32h7xx_hal_flash_ex.c
Compile [ 87.2%]: stm32h7xx_hal_gpio.c
Compile [ 87.3%]: stm32h7xx_hal_hash.c
Compile [ 87.4%]: stm32h7xx_hal_hash_ex.c
Compile [ 87.5%]: stm32h7xx_hal_hcd.c
Compile [ 87.6%]: stm32h7xx_hal_hsem.c
Compile [ 87.7%]: stm32h7xx_hal_i2c_ex.c
Compile [ 87.8%]: stm32h7xx_hal_i2s_ex.c
Compile [ 87.9%]: stm32h7xx_hal_i2s.c
Compile [ 88.1%]: stm32h7xx_hal_hrtim.c
Compile [ 88.2%]: stm32h7xx_hal_i2c.c
Compile [ 88.3%]: stm32h7xx_hal_iwdg.c
Compile [ 88.4%]: stm32h7xx_hal_irda.c
Compile [ 88.5%]: stm32h7xx_hal_jpeg.c
Compile [ 88.6%]: stm32h7xx_hal_lptim.c
Compile [ 88.7%]: stm32h7xx_hal_ltdc_ex.c
Compile [ 88.8%]: stm32h7xx_hal_ltdc.c
Compile [ 88.9%]: stm32h7xx_hal_mdios.c
Compile [ 89.0%]: stm32h7xx_hal_mmc_ex.c
Compile [ 89.1%]: stm32h7xx_hal_mdma.c
Compile [ 89.2%]: stm32h7xx_hal_mmc.c
Compile [ 89.3%]: stm32h7xx_hal_nor.c
Compile [ 89.4%]: stm32h7xx_hal_nand.c
Compile [ 89.5%]: stm32h7xx_hal_opamp.c
Compile [ 89.6%]: stm32h7xx_hal_opamp_ex.c
Compile [ 89.7%]: stm32h7xx_hal_ospi.c
Compile [ 89.8%]: stm32h7xx_hal_otfdec.c
Compile [ 89.9%]: stm32h7xx_hal_pcd_ex.c
Compile [ 90.0%]: stm32h7xx_hal_pcd.c
Compile [ 90.1%]: stm32h7xx_hal_pssi.c
Compile [ 90.3%]: stm32h7xx_hal_pwr.c
Compile [ 90.4%]: stm32h7xx_hal_ramecc.c
Compile [ 90.5%]: stm32h7xx_hal_pwr_ex.c
Compile [ 90.6%]: stm32h7xx_hal_qspi.c
Compile [ 90.7%]: stm32h7xx_hal_rcc.c
Compile [ 90.8%]: stm32h7xx_hal_rng.c
Compile [ 90.9%]: stm32h7xx_hal_rng_ex.c
Compile [ 91.0%]: stm32h7xx_hal_rcc_ex.c
Compile [ 91.1%]: stm32h7xx_hal_rtc.c
Compile [ 91.2%]: stm32h7xx_hal_sai_ex.c
Compile [ 91.3%]: stm32h7xx_hal_rtc_ex.c
Compile [ 91.4%]: stm32h7xx_hal_sai.c
Compile [ 91.5%]: stm32h7xx_hal_sd_ex.c
Compile [ 91.6%]: stm32h7xx_hal_sdram.c
Compile [ 91.7%]: stm32h7xx_hal_sd.c
Compile [ 91.8%]: stm32h7xx_hal_smartcard_ex.c
Compile [ 91.9%]: stm32h7xx_hal_smartcard.c
Compile [ 92.0%]: stm32h7xx_hal_spdifrx.c
Compile [ 92.1%]: stm32h7xx_hal_smbus.c
Compile [ 92.2%]: stm32h7xx_hal_spi_ex.c
Compile [ 92.3%]: stm32h7xx_hal_sram.c
Compile [ 92.5%]: stm32h7xx_hal_spi.c
Compile [ 92.6%]: stm32h7xx_hal_swpmi.c
Compile [ 92.7%]: stm32h7xx_hal_tim_ex.c
Compile [ 92.8%]: stm32h7xx_hal_uart_ex.c
Compile [ 92.9%]: stm32h7xx_hal_tim.c
Compile [ 93.0%]: stm32h7xx_hal_uart.c
Compile [ 93.1%]: stm32h7xx_hal_usart_ex.c
Compile [ 93.2%]: stm32h7xx_hal_wwdg.c
Compile [ 93.3%]: stm32h7xx_hal_usart.c
Compile [ 93.4%]: stm32h7xx_ll_bdma.c
Compile [ 93.5%]: stm32h7xx_ll_comp.c
Compile [ 93.6%]: stm32h7xx_ll_adc.c
Compile [ 93.7%]: stm32h7xx_ll_crc.c
Compile [ 93.8%]: stm32h7xx_ll_crs.c
Compile [ 93.9%]: stm32h7xx_ll_delayblock.c
Compile [ 94.0%]: stm32h7xx_ll_dac.c
Compile [ 94.1%]: stm32h7xx_ll_dma.c
Compile [ 94.2%]: stm32h7xx_ll_fmc.c
Compile [ 94.3%]: stm32h7xx_ll_exti.c
Compile [ 94.4%]: stm32h7xx_ll_dma2d.c
Compile [ 94.5%]: stm32h7xx_ll_gpio.c
Compile [ 94.7%]: stm32h7xx_ll_hrtim.c
Compile [ 94.8%]: stm32h7xx_ll_i2c.c
Compile [ 94.9%]: stm32h7xx_ll_lptim.c
Compile [ 95.0%]: stm32h7xx_ll_lpuart.c
Compile [ 95.1%]: stm32h7xx_ll_opamp.c
Compile [ 95.2%]: stm32h7xx_ll_mdma.c
Compile [ 95.3%]: stm32h7xx_ll_pwr.c
Compile [ 95.4%]: stm32h7xx_ll_rng.c
Compile [ 95.5%]: stm32h7xx_ll_rtc.c
Compile [ 95.6%]: stm32h7xx_ll_rcc.c
Compile [ 95.7%]: stm32h7xx_ll_sdmmc.c
Compile [ 95.8%]: stm32h7xx_ll_spi.c
Compile [ 95.9%]: stm32h7xx_ll_swpmi.c
Compile [ 96.0%]: stm32h7xx_ll_usart.c
Compile [ 96.1%]: stm32h7xx_ll_tim.c
Compile [ 96.2%]: system_stm32h7xx_dualcore_boot_cm4_cm7.c
Compile [ 96.3%]: stm32h7xx_ll_utils.c
Compile [ 96.4%]: system_stm32h7xx_singlecore.c
Compile [ 96.5%]: startup_stm32h747xx.S
Compile [ 96.6%]: stm32h7xx_ll_usb.c
...
```

closed 2020-06-29 by JanneKiiskila