ARMmbed/mbed-os #11606: Cellular: Cannot use private "_is_connected" flag.

mudassar-ublox, 2019-10-01:
In cellular api's [_is_connected](https://github.com/ARMmbed/mbed-os/blob/c385e141e2c2155e92d88cd99fc0f675d900d102/features/cellular/framework/AT/AT_CellularContext.h#L123) flag is private and is used in [do_connect](https://github.com/ARMmbed/mbed-os/blob/c385e141e2c2155e92d88cd99fc0f675d900d102/features/cellular/framework/AT/AT_CellularContext.h#L87) function and `do_connect `is virtual. How child classes can use this flag, `_is_connected`. when `do_connect `is overridden child classes.

Also when `do_connect ` is overridden then we cannot set this flag, `_is_connected `. So while calling [do_disconnect](https://github.com/ARMmbed/mbed-os/blob/c385e141e2c2155e92d88cd99fc0f675d900d102/features/cellular/framework/AT/AT_CellularContext.cpp#L658) because of `_is_connected ` [this ](https://github.com/ARMmbed/mbed-os/blob/c385e141e2c2155e92d88cd99fc0f675d900d102/features/cellular/framework/AT/AT_CellularContext.cpp#L660)check become `true ` and sets `_cid = -1` and when [deactivate_context ](https://github.com/ARMmbed/mbed-os/blob/c385e141e2c2155e92d88cd99fc0f675d900d102/features/cellular/framework/AT/AT_CellularContext.cpp#L731)function is called give error. 

[INFO][CELL]: AT TX (11): AT+CGACT=0,
[INFO][CELL]: AT TX ( 2): -1
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 9): <ln>ERROR

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

mudassar-ublox, 2019-10-01:
@wajahat-ublox @fahim-ublox

0xc0170, 2019-10-01:
cc @ARMmbed/mbed-os-wan

kivaisan, 2019-11-05:
@ARMmbed/mbed-os-maintainers , I think this issue can be closed. Fix has been merged a long time ago.

closed 2019-11-05 by 0xc0170