ARMmbed/mbed-os #11608: STM32H747I - serial stress test - loses characters

JanneKiiskila, 2019-10-01:
### Description

Using current tip of the master of Mbed OS (`mbed.os.lib`)
https://github.com/ARMmbed/mbed-os/#31114ba3436014619d157beadde7a5294f10f54b

Using the following `main.cpp`

```
#include "mbed.h"

Serial pc(USBTX, USBRX);

#define ASCII_SPACE 0x20
#define ASCII_TILDE 0x7f

int main() {

    int c = ASCII_SPACE;

    while(1) {
        c++;
        pc.printf("%c", c);

        if (c == ASCII_TILDE) {
            c = ASCII_SPACE;
            printf("\n");
            pc.fsync();
        }
    }
}
```

and following `mbed_app.json`

```
{
    "target_overrides": {
        "*": {
            "platform.stdio-baud-rate"                  : 9600,
            "platform.stdio-convert-newlines"           : true,
            "platform.stdio-buffered-serial"            : true,
            "platform.stdio-flush-at-exit"              : true
        }
    }
}
```

The serial is losing (in some kind of pattern) the 1-2 first characters of the line (! or !").

```
!"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
!"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
!"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
!"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
```

All the lines should be the same, but some characters are missing in real life.

### Issue request type
    [ ] Question
    [ ] Enhancement
    [x] Bug

JanneKiiskila, 2019-10-01:
NXP's K64F fares significantly better on this, though it's not perfect either (it seems to lose the 1st character but only every 50 or so lines).

JanneKiiskila, 2019-10-02:
I pushed the app to my fork of Mbed-OS; https://github.com/JanneKiiskila/mbed-os/tree/serial-testapp/TEST_APPS/device/serial-testapp

LMESTM, 2019-10-02:
Is the last char supposed to be there : � ? 
I think it shouldn't .. which may mean that the issue is maybe due to sending end of chars. 
Does the same happen without call to: pc.fsync();  ?
What is specific to H7 is only that this STM32 processor is part of the High Performance family and basically faster than other ... which may explain why you see this more often.

JanneKiiskila, 2019-10-02:
The output should be limited by the baud rate, rather than the CPU. Power of the CPU should not impact it, IMHO.

LMESTM, 2019-10-02:
Only my guess ... but if the CPU is faster it may "reset" or act on the Serial HW when calling pc.fsync(); and prevent the end of the line to be sent completely and mess up with the next new line ... that's the reason for my other question. The pc.fsync() doesn't seem necessary to me  or is it ?

LMESTM, 2019-10-02:
BTW - what toolchain ?

JanneKiiskila, 2019-10-02:
I am leaning forward a bit, I'm already using `gcc-arm-none-eabi-8-2019-q3-update`.

JanneKiiskila, 2019-10-02:
I added checks for the status of `pc.fsync()` - it does not return errors. So, it claims to flush OK. The line ending is probably just a CR/LF incompatibility from the copy paste to here (`putty.log` stored, copy pasted the lines from there). The putty log itself looks OK.

I removed the `pc.fsync()` completely, situation is the same nevertheless - characters are being lost from the beginning of the lines.

LMESTM, 2019-10-02:
Ok I ended up reproducing the issue here and I figured out your mixing up with 2 different objects in your application: Serial pc and the default stdio serial object.
The default stdio is being used by direct calls to printf() (not pc.printf).
I could make the application work ok by moving line
`            printf("\n");`
by 
`            pc.printf("\n");`

JanneKiiskila, 2019-10-03:
Douh, stupid user error then.

closed 2019-10-03 by JanneKiiskila