ARMmbed/mbed-os #11610: Support for STM32G0 and STM32G4

martinjaeger, 2019-10-02:
### Description

Is it planned to support the new STM32G series in the near future?

I'm particularly interested in the G4 series, as I want to use the STM32G431 for a new product development.

On [ST website](https://www.st.com/en/evaluation-tools/nucleo-g431rb.html) they state that the Nucleo G431RB is already mbed compliant, but can't find it under `targets`.

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

jeromecoutant, 2019-10-02:
@ARMmbed/team-st-mcd

closed 2019-10-25 by ciarmcom

martinjaeger, 2019-10-25:
Mhh, not very nice to just close without any answer even though this issue was raised before the policy changed.

martinjaeger, 2019-10-25:
By the way, the same question was [raised in the forum already](https://forums.mbed.com/t/new-board-info-for-stm32g0-series/5531) for STM32G0 without any answer since August 30.

0xc0170, 2019-10-28:
> Mhh, not very nice to just close without any answer even though this issue was raised before the policy changed.

You still can get an answer. As we don't have an answer to all questions, some might be without an answer unfortunately. 
True, the policy has an affect now for older issues. Those relevant can be opened on the forum and just comment here,  we can still look at some.

@ARMmbed/team-st-mcd Would you be able to provide an answer for this question?

0xc0170, 2019-10-28:
> By the way, the same question was raised in the forum already for STM32G0 without any answer since August 30.

@NadyaARM

LMESTM, 2019-10-28:
sorry for not answering. We're looking at all requests and trying to prioritize targets roadmap. 
We'll let you know when a plan is confirmed for G0 and/or G4.

martinjaeger, 2019-10-28:
Awesome. Thanks. I have already started porting parts of my firmware to Zephyr, but would appreciate to be able to stay with mbed for now.

Didn't know that specific link for new STM32 platforms. Unfortunately, I failed to get a not supported STM32L452 Nucleo board running with mbed some time ago (probably because of some linker file issues), so I'd rather not try to port an entirely new family of MCUs.

jeromecoutant, 2019-12-12:
Hi all
For your information, STM32G0 family has been introduced in #12093 
Regards,

jeromecoutant, 2020-02-18:
Hi

For STM32G4, I can share the results of "automatic" script results in
https://github.com/jeromecoutant/mbed/tree/DEV_STM32G4

This branch can't compile, but I think it is a good starting point.

Regards,
Jerome

martinjaeger, 2020-02-18:
Hi Jerome, thanks a lot for your effort.

I decided to migrate my firmware to Zephyr RTOS. But I'm sure the STM32G4 support will be helpful for others as well.

AGlass0fMilk, 2020-02-20:
@jeromecoutant 
I took your branch and have basic support for the NUCLEO_G747RE target (gpio and system initialization, enough to compile blinky while ignoring a lot of features that require UART).

You can see my branch here: https://github.com/AGlass0fMilk/mbed-os/tree/add-nucleo-g474re

I am hoping we can collaborate a bit to add the rest of the features since you are obviously more familiar with the STM platform than I am.

Maybe you could convince your project managers to give you some time to assist me since there is another developer working on this now :)?

jeromecoutant, 2020-02-20:
@AGlass0fMilk 
Please push a pull request for official review with all interested people.
No problem to enable feature 1 by 1, as long as you check tests results.

AGlass0fMilk, 2020-02-20:
See #12482

CRA-CNRS, 2021-11-04:
Hi what's new about the arm Mbed support of Nucleo G431RB ? The ST website says that the board is Mbed compliant, while in the datasheet it says the countrary. Who is right ?

The st board: https://www.st.com/en/evaluation-tools/nucleo-g431rb.html
The datasheet: (click on download databrief to see the datasheet)

jeromecoutant, 2021-11-08:
> Hi what's new about the arm Mbed support of Nucleo G431RB and Nucleo G474RE? The ST website says that the board is Mbed compliant, while in the datasheet it says the countrary. Who is right ?

Nucleo G474RE is supported:
https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L3011-L3037

Nucleo G431RB as well:
https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L2935-L2961

CRA-CNRS, 2021-11-08:
@jeromecoutant  thank you but somebody should update the documentation. (link here: https://www.st.com/en/evaluation-tools/nucleo-g431rb.html)

[image]