ARMmbed/mbed-os #11611: STM32H747I - mbed-bootloader does not compile (lack of MBED_ROM -defines)

JanneKiiskila, 2019-10-02:
### Description

Target:
- `DISCO_H747I`

Toolchain (name + version) are you using?
- GCC_ARM, doubt the compiler makes a difference here.

- Steps to reproduce:

```
git clone -b DISCO_H747I git@github.com:JanneKiiskila/mbed-bootloader.git
mbed compile -t GCC_ARM -m DISCO_H747I --profile=tiny.json --app-config=configs/internal_flash_no_rot.json
```

As a reference - this will compile:

```
mbed compile -t GCC_ARM -m DISCO_L475VG_IOT01A --profile=tiny.json --app-config=configs/internal_flash_no_rot.json
```

However, the H747 will fail as the MBED_ROM -defines are missing for some reason.

Error log:

```
Compile [ 99.2%]: arm_uc_pal_flashiap_implementation.c
[Error] mbed_config.h@92,76: 'MBED_ROM_START' undeclared (first use in this function)
[Error] mbed_config.h@114,75: 'MBED_ROM_SIZE' undeclared (first use in this function); did you mean 'MFST_ERR_SIZE'?
[Error] mbed_config.h@89,76: 'MBED_ROM_START' undeclared (first use in this function)
[Error] mbed_config.h@38,75: 'APPLICATION_SIZE' undeclared (first use in this function); did you mean 'ARM_UC_ROT_SIZE'?
[ERROR] In file included from <command-line>:
./modules/storage/pal-flashiap/source/arm_uc_pal_flashiap_implementation.c: In function 'arm_uc_pal_flashiap_get_slot_addr_size':
././BUILD/DISCO_H747I/GCC_ARM-TINY/mbed_config.h:92:76: error: 'MBED_ROM_START' undeclared (first use in this function)
 #define MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS                           (MBED_ROM_START + MBED_BOOTLOADER_FLASH_BANK_SIZE + KVSTORE_SIZE)                                        // set by application[*]
                                                                            ^~~~~~~~~~~~~~
./modules/storage/pal-flashiap/source/arm_uc_pal_flashiap_implementation.c:106:39: note: in expansion of macro 'MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS'
                                       MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS, 0);
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
././BUILD/DISCO_H747I/GCC_ARM-TINY/mbed_config.h:92:76: note: each undeclared identifier is reported only once for each function it appears in
 #define MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS                           (MBED_ROM_START + MBED_BOOTLOADER_FLASH_BANK_SIZE + KVSTORE_SIZE)                                        // set by application[*]
                                                                            ^~~~~~~~~~~~~~
./modules/storage/pal-flashiap/source/arm_uc_pal_flashiap_implementation.c:106:39: note: in expansion of macro 'MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS'
                                       MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS, 0);
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
././BUILD/DISCO_H747I/GCC_ARM-TINY/mbed_config.h:114:75: error: 'MBED_ROM_SIZE' undeclared (first use in this function); did you mean 'MFST_ERR_SIZE'?
 #define MBED_BOOTLOADER_FLASH_BANK_SIZE                                   MBED_ROM_SIZE/2                                                                                          // defined by application
                                                                           ^~~~~~~~~~~~~
././BUILD/DISCO_H747I/GCC_ARM-TINY/mbed_config.h:92:93: note: in expansion of macro 'MBED_BOOTLOADER_FLASH_BANK_SIZE'
 #define MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS                           (MBED_ROM_START + MBED_BOOTLOADER_FLASH_BANK_SIZE + KVSTORE_SIZE)                                        // set by application[*]
                                                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./modules/storage/pal-flashiap/source/arm_uc_pal_flashiap_implementation.c:106:39: note: in expansion of macro 'MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS'
                                       MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS, 0);
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./modules/storage/pal-flashiap/source/arm_uc_pal_flashiap_implementation.c: In function 'ARM_UC_PAL_FlashIAP_GetActiveDetails':
././BUILD/DISCO_H747I/GCC_ARM-TINY/mbed_config.h:89:76: error: 'MBED_ROM_START' undeclared (first use in this function)
 #define MBED_CONF_UPDATE_CLIENT_APPLICATION_DETAILS                       (MBED_ROM_START + MBED_BOOTLOADER_SIZE)                                                                  // set by application[*]
                                                                            ^~~~~~~~~~~~~~
./modules/storage/pal-flashiap/source/arm_uc_pal_flashiap_implementation.c:508:13: note: in expansion of macro 'MBED_CONF_UPDATE_CLIENT_APPLICATION_DETAILS'
         if (MBED_CONF_UPDATE_CLIENT_APPLICATION_DETAILS) {
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
././BUILD/DISCO_H747I/GCC_ARM-TINY/mbed_config.h:38:75: error: 'APPLICATION_SIZE' undeclared (first use in this function); did you mean 'ARM_UC_ROT_SIZE'?
 #define MBED_BOOTLOADER_SIZE                                              APPLICATION_SIZE                                                                                         // set by application[*]
                                                                           ^~~~~~~~~~~~~~~~
././BUILD/DISCO_H747I/GCC_ARM-TINY/mbed_config.h:89:93: note: in expansion of macro 'MBED_BOOTLOADER_SIZE'
 #define MBED_CONF_UPDATE_CLIENT_APPLICATION_DETAILS                       (MBED_ROM_START + MBED_BOOTLOADER_SIZE)                                                                  // set by application[*]
                                                                                             ^~~~~~~~~~~~~~~~~~~~
./modules/storage/pal-flashiap/source/arm_uc_pal_flashiap_implementation.c:508:13: note: in expansion of macro 'MBED_CONF_UPDATE_CLIENT_APPLICATION_DETAILS'
         if (MBED_CONF_UPDATE_CLIENT_APPLICATION_DETAILS) {
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

[mbed] ERROR: "/usr/bin/python" returned error.
       Code: 1
       Path: "/home/jankii01/mbed/mbed-bootloader-internal"
       Command: "/usr/bin/python -u /home/jankii01/mbed/mbed-bootloader-internal/mbed-os/tools/make.py -t GCC_ARM -m DISCO_H747I --profile tiny.json --source . --build ./BUILD/DISCO_H747I/GCC_ARM-TINY --app-config configs/internal_flash_no_rot.json"
```

There is now some incompatibility in target vs. the defines created by the tooling in Mbed OS. The target claims to support bootloader, so it should work.
- ref: https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L3367

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

JanneKiiskila, 2019-10-02:
Root cause is likely outdated CMSIS-pack, where the defines apparently are.

VeijoPesonen, 2019-10-02:
[CMSIS pack data](https://raw.githubusercontent.com/ARMmbed/mbed-os/master/tools/arm_pack_manager/index.json) - The target is not found from that file.

jeromecoutant, 2019-10-02:
You have to update pack data with the latest pack:
https://www.keil.com/dd2/pack/

JanneKiiskila, 2019-10-02:
Yes, this file does not have H747 at all.

https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json

jeromecoutant, 2019-10-10:
Waiting for #11666 correction

JanneKiiskila, 2019-10-16:
We could probably live for a while by having the values come via `targets/targets.json`.

MarceloSalazar, 2019-10-17:
@jeromecoutant as workaround do you think you could introduce the values in `targets.json` for the H747?
```
"mbed_rom_start"    : "0xxxxx",
"mbed_rom_size"     : "0xxxxx",
```

JanneKiiskila, 2019-10-17:
Unfortunately not, next item the bootloader compilation fails to is then `device_name` missing and then when you put that in place, it still wants the cmsis-pack matching the `device_name`.

JanneKiiskila, 2019-11-06:
This is now sorted out with the CMSIS_pack update. Closing out.

closed 2019-11-06 by JanneKiiskila