ARMmbed/mbed-os #11617: STM32H747I - DeviceKey is not enabled

JanneKiiskila, 2019-10-03:
### Description

Target: DISCO_H747I 
Toolchain: GCC_ARM
Tools: Mbed CLI 1.10.1  
Mbed OS hash: 31114ba3436014619d157beadde7a5294f10f54b

Try to run device key tests as per Mbed OS documents.

`mbed test -n features-device_key-tests-device_key-functionality -m DISCO_H747I -t GCC_ARM`

Docs in: https://os.mbed.com/docs/mbed-os/v5.14/apis/devicekey.html

Error log:

```
Building project functionality (DISCO_H747I, GCC_ARM)
Scan: GCC_ARM
Scan: functionality
Compile [100.0%]: main.cpp
[Error] main.cpp@31,2: #error [NOT_SUPPORTED] DeviceKey needs to be enabled for this test
Memory map breakdown for built projects (values in Bytes):
| name | target | toolchain | static_ram | total_flash |
|------|--------|-----------|------------|-------------|

Build successes:
  * DISCO_H747I::GCC_ARM::MBED-BUILD

Build skips:
  * DISCO_H747I::GCC_ARM::FEATURES-DEVICE_KEY-TESTS-DEVICE_KEY-FUNCTIONALITY
[1570093741.66][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt
[1570093741.71][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
mbedgt: greentea test automation tool ver. 1.6.5
mbedgt: test specification file './BUILD/tests/DISCO_H747I/GCC_ARM/test_spec.json' (specified with --test-spec option)
mbedgt: using './BUILD/tests/DISCO_H747I/GCC_ARM/test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
[1570093741.87][mbedls.lstools_linux]Could not get serial devices by id. This could be because your Linux distribution does not use udev, or does not create /dev/serial/by-id symlinks. Please submit an issue to github.com/armmbed/mbed-ls.
mbedgt: no compatible devices detected
mbedgt: completed in 0.07 sec
mbedgt: exited with code 1
[mbed] ERROR: "mbedgt" returned error.
       Code: 1
       Path: "/home/jankii01/mbed/mbed-cloud-client-example-internal/mbed-os"
       Command: "mbedgt --test-spec ./BUILD/tests/DISCO_H747I/GCC_ARM/test_spec.json -n features-device_key-tests-device_key-functionality"
       Tip: You could retry the last command with "-v" flag for verbose output

```

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

Attn: @jeromecoutant  @ARMmbed/team-st-mcd

LMESTM, 2019-10-03:
@JanneKiiskila 
Is there an example of a target that has this feature enabled by default ?  and what are the requirements for a target to enable this, I don't see a description of what a target needs to implement (HAL layer ?) to support this feature ... can you clarify ?

If there is no HW dependency, then this can rather be enabled from mbed_app.json I think ...

JanneKiiskila, 2019-10-03:
Well, plenty of other ST boards have it enabled - like F411RE, F429ZI, UBLOX_EVK_ODIN_W2.

JanneKiiskila, 2019-10-03:
I would venture a guess that it needs 

```
       "components_add": ["FLASHIAP"],
```

added to the `targets.json`.

PR available; https://github.com/ARMmbed/mbed-os/pull/11619

But I can't test, as the board is now refusing to connect over USB to Windows. Is there any way to rescue boot it or something similar? I could then run the relevant tests.

SeppoTakalo, 2019-10-04:
This is more complicated than just enabling the feature.
The API documentation does not help https://os.mbed.com/docs/mbed-os/v5.14/apis/devicekey.html as it only tells how to use the key. Does not tell how to set it up.

If short, devicekey requires flash, as it is stored inside KVStore. But if the location of KVStore is not set, then there cannot be devicekey.

But the OS cannot set the location of KVStore, as it is application dependant. Application may choose to split the flash in half, and use other part for FW update purposes, or it might use external storage for that.
Therefore any default locations that we give, are wrong for some use cases.

MarceloSalazar, 2019-10-04:
Thanks Seppo for the reasoning.
I agree there might be multiple ways to setup the memory layout depending on the application or certain optimizations.

But can we facilitate a way to provide the *default* config in targets.json (or somewhere else) that would work out of the box with most applications, such Pelion client.
The developer should be able to override it as they see fit.

JanneKiiskila, 2019-10-04:
Yep, we can have a sensible default - that passes for example the Mbed GreenTea tests. Defaulting to store KVStore at the end of the internal flash would be a quite safe assumption. Override is of course needed, as for this board for example, I would assume the following memory layout.

```
bank 1 - erase sector    0 - 128 kB - Bootloader
bank 1 - erase sector 1..7 - 7 * 128 kB - application
bank 2 - erase sector 0..1 - 2 * 128 kB - KVStore
bank 2 - erase sector 2..7 - 6 * 128 kB - FW Download area
```

We probably also need to set the default size somewhere, because KVStore needs to be 2 flash erase sectors. The **erase sector size is non-standard** and especially ST board seem to have very many different sizes. In this particular board, the erase sectors are extremely large - 128 kilobytes each. Just out of curiosity - is there a special reason why they are so large?

JanneKiiskila, 2019-10-04:
By they way, I did add this

```
(venv) jankii01@ubuntu:~/mbed/mbed-os-devicekey/mbed-os$ git diff
diff --git i/features/storage/kvstore/conf/tdb_internal/mbed_lib.json w/features/storage/kvstore/conf/tdb_internal/mbed_lib.json
index 04e1f102ed..e123465bb7 100644
--- i/features/storage/kvstore/conf/tdb_internal/mbed_lib.json
+++ w/features/storage/kvstore/conf/tdb_internal/mbed_lib.json
@@ -15,6 +15,10 @@
             "internal_size": "0x8000",
             "internal_base_address": "0x00028000"
         },
+        "DISCO_H747": {
+            "internal_size": "256*1024*1024",
+            "internal_base_address": "0x08100000"
+        },
```

But it still fails the test.

jeromecoutant, 2019-10-04:
Note that H747 is a dual core chip, see #11605 
Basically:
- M7 core is using bank 1
- M4 core is using bank 2

@ABOSTM

JanneKiiskila, 2019-10-04:
Oh. That's going to be a challenge then, this is normal memory in the sense that if you do sector erase OR you write to a flash bank, it prevents simultaneous code execution? Is there no way to configure the M4 to use SOME sectors from bank2?

This should be documented somewhere.

JanneKiiskila, 2019-10-04:
Related PR by the way; https://github.com/ARMmbed/mbed-os/pull/11629/files - needs review/feedback from all chip vendors.

JanneKiiskila, 2019-10-04:
Got this far now:

```
mbedgt: test on hardware with target id: 081402210248650C3D0CFC25
mbedgt: test suite 'mbed-os-features-storage-tests-kvstore-tdbstore_whitebox' ........................ OK in 10.67 sec
	test case: 'TDBStore: Error inject test' ..................................................... OK in 0.11 sec
	test case: 'TDBStore: Multiple set test' ..................................................... OK in 0.11 sec
	test case: 'TDBStore: White box test' ........................................................ OK in 0.10 sec
mbedgt: test case summary: 3 passes, 0 failures
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.2698047915
mbedgt: test suite report:
| target            | platform_name | test suite                                                          | result | elapsed_time (sec) | copy_method |
|-------------------|---------------|---------------------------------------------------------------------|--------|--------------------|-------------|
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-direct_access_devicekey_test | FAIL   | 19.71              | default     |
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-static_tests                 | OK     | 29.59              | default     |
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-tdbstore_whitebox            | OK     | 10.67              | default     |
mbedgt: test suite results: 1 FAIL / 2 OK
mbedgt: test case report:
| target            | platform_name | test suite                                                          | test case                                                           | passed | failed | result | elapsed_time (sec) |
|-------------------|---------------|---------------------------------------------------------------------|---------------------------------------------------------------------|--------|--------|--------|--------------------|
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-direct_access_devicekey_test | Testing direct access to devicekey with tdb over flashiap remainder | 1      | 0      | OK     | 2.94               |
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-direct_access_devicekey_test | Testing direct access to devicekey with tdb over last two sectors   | 0      | 1      | FAIL   | 2.98               |
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-direct_access_devicekey_test | Testing direct access to injected devicekey                         | 1      | 0      | OK     | 2.85               |
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-static_tests                 | get_buffer_null_size_not_zero                                       | 1      | 0      | OK     | 0.06               |
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-static_tests                 | get_buffer_size_bigger_than_data_real_size                          | 1      | 0      | OK     | 0.07               |
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-static_tests                 | get_buffer_size_is_zero                                             | 1      | 0      | OK     | 0.05               |
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-static_tests                 | get_buffer_size_smaller_than_data_real_size                         | 1      | 0      | OK     | 0.08               |
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-static_tests                 | get_info_existed_key                                                | 1      | 0      | OK     | 1.88               |
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-static_tests                 | get_info_info_null                                                  | 1      | 0      | OK     | 0.04               |
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-static_tests                 | get_info_key_length_exceeds_max                                     | 1      | 0      | OK     | 0.07               |
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-static_tests                 | get_info_key_null                                                   | 1      | 0      | OK     | 0.05               |
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-static_tests                 | get_info_non_existing_key                                           | 1      | 0      | OK     | 0.06               |
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-static_tests                 | get_info_overwritten_key                                            | 1      | 0      | OK     | 0.05               |
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-static_tests                 | get_info_removed_key                                                | 1      | 0      | OK     | 0.05               |
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-static_tests                 | get_key_length_exceeds_max                                          | 1      | 0      | OK     | 0.05               |
[... cut ...]
| DISCO_H747I-ARMC6 | DISCO_H747I   | mbed-os-features-storage-tests-kvstore-tdbstore_whitebox            | TDBStore: White box test                                            | 1      | 0      | OK     | 0.1                |
mbedgt: test case results: 1 FAIL / 54 OK
mbedgt: completed in 60.35 sec
```

So, only one test failing. I wonder if the `mbed-os-features-storage-tests-kvstore-direct_access_devicekey_test` actually makes some hard assumptions, that fail with this special HW? Is the bank2 out of range, even you run it in single-core mode?

JanneKiiskila, 2019-10-04:
Yep, that is the case. Adding some prints there it's easy to see, that.

https://github.com/ARMmbed/mbed-os/blob/2e09a27e05dfd412ebef6786e5736fcf0a906d98/features/storage/TESTS/kvstore/direct_access_devicekey_test/main.cpp#L180-L184

If you add printouts there, it will give you this;

```
[1570191918.98][CONN][RXD] Test Direct Access To DeviceKey Test Last Two Sectors
[1570191919.01][CONN][RXD] flash_bd_start_address = 0x81c0000
[1570191919.05][CONN][RXD] flash_bd_end_address = 0x8200000
[1570191919.08][CONN][RXD] flash_bd_size = 0x40000
```

Which places this test to be executed with KVStore being in the last two erase sectors of bank 2, which the CPU in this case can't access? Now the bigger philosophical/architectural question is - how on earth should this be handled? Should the flash.h file be changed, so that for this target - the boards should only list 1 bank with size of 1 megabyte, instead of the true HW? Then the other M4 target will see only the bank 2 with size of 1 megabyte?

Having a target specific "oh, with this board I'll drop it to end of bank1 in test case" will just be and endless nightmare going forwards...

@bulislaw

jeromecoutant, 2019-10-07:
> Should the flash.h file be changed, so that for this target - the boards should only list 1 bank with size of 1 megabyte, instead of the true HW?

Each core has write access to each bank.
We didn't want to avoid one core to write something for the other core side.

JanneKiiskila, 2019-10-07:
I would believe the test would pass, if you switch the bank access around.
- 1st MB for M4 only.
- 2nd MB for M7, that way the KVSTore/TDBStore would also run in the end of the bank.
- Just changing the [flash size to 1 MB](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32H7/TARGET_STM32H747xI/flash_data.h#L30) does not for some reason push the TDBtore to end of the 1st bank, there is some more complicated logic how it calculates that.

jeromecoutant, 2019-10-08:
So next step is patching TDBStore to allow to use some specific memory part, and not the end of flash?

JanneKiiskila, 2019-10-08:
Next step is to come up with a sensible way to tell each CPU, what is the end of the addressable/writeable flash from their point of view.

JanneKiiskila, 2019-10-08:
The current TDBStore tests gets the placing via the [FlashIAP.h](https://github.com/ARMmbed/mbed-os/blob/master/drivers/FlashIAP.h),  it is using these;

https://github.com/ARMmbed/mbed-os/blob/master/drivers/FlashIAP.h#L38-L47
i.e.  `FLASHIAP_APP_ROM_END_ADDR` and then starts iterating downwards from that, until it can fit the store size to it.

JanneKiiskila, 2019-10-31:
I think this can be closed, https://github.com/ARMmbed/mbed-os/pull/11619 fixes it.

jeromecoutant, 2019-10-31:
Confirmed, to be closed thanks to #11619

closed 2019-10-31 by ciarmcom