ARMmbed/mbed-os #11633: QSPIF general_block_device test is failing on EFM32GG11_STK3701

maciejbocianski, 2019-10-04:
### Description

`QSPIF Testing program read small data sizes` test is failing on `EFM32GG11_STK3701`

GCC_ARM + master

```
[1570191655.44][CONN][RXD] >>> Running case #6: 'QSPIF Testing program read small data sizes'...
[1570191655.44][CONN][RXD]
[1570191655.44][CONN][RXD] Test program-read small data sizes, from 1 to 7 bytes..
[1570191655.44][CONN][INF] found KV pair in stream: {{__testcase_start;QSPIF Testing program read small data sizes}}, queued...
mbedgt: :614::FAIL: Expected 0 Was -4001
[1570191655.52][CONN][RXD] :614::FAIL: Expected 0 Was -4001
[1570191655.52][CONN][RXD] >>> failure with reason 'No Failure' during 'Unknown Location'
mbedgt: :623::FAIL: Expected 0 Was 19
[1570191655.52][CONN][RXD] :623::FAIL: Expected 0 Was 19
[1570191655.52][CONN][RXD] >>> failure with reason 'No Failure' during 'Unknown Location'
mbedgt: :614::FAIL: Expected 0 Was -4001
[1570191655.58][CONN][RXD] :614::FAIL: Expected 0 Was -4001
[1570191655.58][CONN][RXD] >>> failure with reason 'No Failure' during 'Unknown Location'
mbedgt: :620::FAIL: Expected 0 Was -4001
[1570191655.60][CONN][RXD] :620::FAIL: Expected 0 Was -4001
[1570191655.60][CONN][RXD] >>> failure with reason 'No Failure' during 'Unknown Location'
mbedgt: :623::FAIL: Expected 0 Was 19
[1570191655.60][CONN][RXD] :623::FAIL: Expected 0 Was 19
[1570191655.60][CONN][RXD] >>> failure with reason 'No Failure' during 'Unknown Location'
mbedgt: :614::FAIL: Expected 0 Was -4001
[1570191655.67][CONN][RXD] :614::FAIL: Expected 0 Was -4001
[1570191655.67][CONN][RXD] >>> failure with reason 'No Failure' during 'Unknown Location'
mbedgt: :620::FAIL: Expected 0 Was -4001
[1570191655.69][CONN][RXD] :620::FAIL: Expected 0 Was -4001
[1570191655.69][CONN][RXD] >>> failure with reason 'No Failure' during 'Unknown Location'
mbedgt: :623::FAIL: Expected 0 Was 19
[1570191655.69][CONN][RXD] :623::FAIL: Expected 0 Was 19
[1570191655.69][CONN][RXD] >>> failure with reason 'No Failure' during 'Unknown Location'
mbedgt: :614::FAIL: Expected 0 Was -4001
[1570191655.83][CONN][RXD] :614::FAIL: Expected 0 Was -4001
[1570191655.83][CONN][RXD] >>> failure with reason 'No Failure' during 'Unknown Location'
mbedgt: :623::FAIL: Expected 0 Was 19
[1570191655.83][CONN][RXD] :623::FAIL: Expected 0 Was 19
[1570191655.83][CONN][RXD] >>> failure with reason 'No Failure' during 'Unknown Location'
mbedgt: :614::FAIL: Expected 0 Was -4001
[1570191655.89][CONN][RXD] :614::FAIL: Expected 0 Was -4001
[1570191655.89][CONN][RXD] >>> failure with reason 'No Failure' during 'Unknown Location'
mbedgt: :620::FAIL: Expected 0 Was -4001
[1570191655.91][CONN][RXD] :620::FAIL: Expected 0 Was -4001
[1570191655.91][CONN][RXD] >>> failure with reason 'No Failure' during 'Unknown Location'
mbedgt: :623::FAIL: Expected 0 Was 19
[1570191655.91][CONN][RXD] :623::FAIL: Expected 0 Was 19
[1570191655.91][CONN][RXD] >>> failure with reason 'No Failure' during 'Unknown Location'
mbedgt: :614::FAIL: Expected 0 Was -4001
[1570191655.97][CONN][RXD] :614::FAIL: Expected 0 Was -4001
[1570191655.97][CONN][RXD] >>> failure with reason 'No Failure' during 'Unknown Location'
mbedgt: :620::FAIL: Expected 0 Was -4001
[1570191655.99][CONN][RXD] :620::FAIL: Expected 0 Was -4001
[1570191655.99][CONN][RXD] >>> failure with reason 'No Failure' during 'Unknown Location'
mbedgt: :623::FAIL: Expected 0 Was 19
[1570191655.99][CONN][RXD] :623::FAIL: Expected 0 Was 19
[1570191656.01][CONN][RXD] >>> failure with reason 'No Failure' during 'Unknown Location'
[1570191656.01][CONN][INF] found KV pair in stream: {{__testcase_finish;QSPIF Testing program read small data sizes;0;16}}, queued...
[1570191656.02][CONN][RXD] >>> 'QSPIF Testing program read small data sizes': 0 passed, 16 failed with reason 'Test Cases Failed'

...

| target                    | platform_name     | test suite                                              | result | elapsed_time (sec) | copy_method |
|---------------------------|-------------------|---------------------------------------------------------|--------|--------------------|-------------|
| EFM32GG11_STK3701-GCC_ARM | EFM32GG11_STK3701 | features-storage-tests-blockdevice-general_block_device | FAIL   | 28.95              | default     |
| target                    | platform_name     | test suite                                              | test case                                      | passed | failed | result | elapsed_time (sec) |
|---------------------------|-------------------|---------------------------------------------------------|------------------------------------------------|--------|--------|--------|--------------------|
| EFM32GG11_STK3701-GCC_ARM | EFM32GG11_STK3701 | features-storage-tests-blockdevice-general_block_device | DEFAULT Testing get type functionality         | 1      | 0      | OK     | 0.0                |
| EFM32GG11_STK3701-GCC_ARM | EFM32GG11_STK3701 | features-storage-tests-blockdevice-general_block_device | QSPIF Testing BlockDevice erase functionality  | 1      | 0      | OK     | 0.18               |
| EFM32GG11_STK3701-GCC_ARM | EFM32GG11_STK3701 | features-storage-tests-blockdevice-general_block_device | QSPIF Testing Deinit block device              | 1      | 0      | OK     | 0.02               |
| EFM32GG11_STK3701-GCC_ARM | EFM32GG11_STK3701 | features-storage-tests-blockdevice-general_block_device | QSPIF Testing Init block device                | 1      | 0      | OK     | 0.04               |
| EFM32GG11_STK3701-GCC_ARM | EFM32GG11_STK3701 | features-storage-tests-blockdevice-general_block_device | QSPIF Testing contiguous erase, write and read | 1      | 0      | OK     | 4.32               |
| EFM32GG11_STK3701-GCC_ARM | EFM32GG11_STK3701 | features-storage-tests-blockdevice-general_block_device | QSPIF Testing multi threads erase program read | 1      | 0      | OK     | 5.87               |
| EFM32GG11_STK3701-GCC_ARM | EFM32GG11_STK3701 | features-storage-tests-blockdevice-general_block_device | QSPIF Testing program read small data sizes    | 0      | 16     | FAIL   | 0.57               |
| EFM32GG11_STK3701-GCC_ARM | EFM32GG11_STK3701 | features-storage-tests-blockdevice-general_block_device | QSPIF Testing read write random blocks         | 1      | 0      | OK     | 1.21               |
| EFM32GG11_STK3701-GCC_ARM | EFM32GG11_STK3701 | features-storage-tests-blockdevice-general_block_device | QSPIF Testing unaligned erase blocks           | 1      | 0      | OK     | 0.06               |
```

@jamesbeyond 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

jamesbeyond, 2019-10-04:
@ARMmbed/team-silabs

stevew817, 2019-10-04:
Well, that's a useful error description...

maciejbocianski, 2019-10-04:
fix: https://github.com/ARMmbed/mbed-os/pull/11636

jamesbeyond, 2019-10-04:
Seems we don't need to worries about description anymore @stevew817

0xc0170, 2019-11-06:
> fix: #11636

Fixed

closed 2019-11-06 by 0xc0170