ARMmbed/mbed-os #11643: Cypress GPIO pull mode tests fail on CY8CKIT_062_WIFI_BT

maclobdell, 2019-10-07:
### Description

Target: CY8CKIT_062_WIFI tested, most likely all PSoC 6 targets affected
Mbed OS ver: 5.14.0
Toolchain: doesn't matter
Hardware: FPGA Test Shield + CY8CKIT_062_WIFI_BT

**Steps to reproduce**
1. Modify the board to remove resistor R100 and add 0 ohm resistor R104 (I just shorted R104 pads together).  This disables the level shifter that connects some gpio/i2c/spi to the debug interface).
2. Attach Mbed FPGA Test shield & Arduino form factor adapter.  Plug in one usb cable to power the shield and one cable to power the board under test.  
3.  Import the latest Mbed OS 5.14.0
4. Add mbed_app.json in the root directory.  Add these contents to enable FPGA test shield tests.

```
{
    "target_overrides": {
        "*": {
            "target.components_add": ["FPGA_CI_TEST_SHIELD"],
        }
    }
}

```

5. Add this function to the end of \targets\TARGET_Cypress\TARGET_PSOC6\mbed_overrides.c.  This avoids testing D2, D14, D15 which have external pull downs or pull ups.  

```
const PinList *pinmap_restricted_pins()
{
	/*D2 has a pull down on board, D14, D15 have pullups on the board */
    static const PinName pins[] = {
        USBRX, USBTX, D2, D14, D15
    };
    static const PinList pin_list = {
        sizeof(pins) / sizeof(pins[0]),
        pins
    };
    return &pin_list;
}
```

6. Run the tests with the following command (after previously setting target and toolchain)

```
mbed test -n tests-mbed_hal_fpga_ci_test_shield-gpio -v

```

**Issues 1**

Testcases in  \TESTS\mbed_hal_fpga_ci_test_shield\gpio\main.cpp

The "tester.gpio_write" will drive the pin high or low.  The gpio_read will check the state of the pin.

The first and third assert will fail. 
```
   // Test input, pull-none mode.
    gpio_mode(&gpio, PullNone);
    tester.gpio_write(MbedTester::LogicalPinGPIO0, 1, true);
    TEST_ASSERT_EQUAL_INT(1, gpio_read(&gpio));
    tester.gpio_write(MbedTester::LogicalPinGPIO0, 0, true);
    TEST_ASSERT_EQUAL_INT(0, gpio_read(&gpio));
    tester.gpio_write(MbedTester::LogicalPinGPIO0, 1, true);
    TEST_ASSERT_EQUAL_INT(1, gpio_read(&gpio));
```
Suspected cause: 
In cy_gpio_api.c, there doesn't seem to be any handling of PullNone.  The expectation is that the pin remains in input mode, but the pullup or pulldown is disabled.   Is there any logic that should be added to explicilty disable the pull resistor?

```
void apply_config(gpio_t *obj)
{
    MBED_ASSERT(obj->pin != CYHAL_NC_PIN_VALUE);

    cy_rslt_t rslt;
    if (CY_RSLT_SUCCESS != (rslt = cyhal_gpio_configure(obj->pin, obj->direction, obj->drive_mode))) {
        MBED_ERROR(MBED_MAKE_ERROR(MBED_MODULE_DRIVER_GPIO, CY_RSLT_GET_CODE(rslt)), "cyhal_gpio_configure failed");
    }

    if (obj->drive_mode == PullUp) {
        gpio_write(obj, 1);
    } else if (obj->drive_mode == PullDown) {
        gpio_write(obj, 0);
    }
}
``` 

**Issue 2**
 gpio_init_inout() fails to enable pull up when requested

Testcases in \TESTS\mbed_hal_fpga_ci_test_shield\gpio\main.cpp

Setting the PullUp, appears to not work in this section of the code.  It is supposed to enable the pullup and the pin should float high, but it doesn't.  
```
    // Initialize GPIO pin as an input, pull-up mode.
    memset(&gpio, 0, sizeof gpio);
    gpio_init_inout(&gpio, pin, PIN_INPUT, PullUp, 0);
    TEST_ASSERT_NOT_EQUAL(0, gpio_is_connected(&gpio));
    tester.gpio_write(MbedTester::LogicalPinGPIO0, 0, false);
    wait_us(HI_Z_READ_DELAY_US);
    TEST_ASSERT_EQUAL_INT(1, gpio_read(&gpio)); // hi-Z, pulled up
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

maclobdell, 2019-10-07:
cc @ARMmbed/team-cypress

0xc0170, 2019-11-06:
Fixed via #11765

closed 2019-11-06 by 0xc0170