ARMmbed/mbed-os #11658: error outputting test results to json with Python 3

maclobdell, 2019-10-09:
## Description

Mbed OS: 5.14.0
toolchain: doesn't matter
python version: 3.7
python package versions: 
    mbed-cli (1.10.0)  
    mbed-clitest (0.21.0)
    mbed-greentea (1.7.1)
    mbed-host-tests (1.5.7)
    mbed-ls (1.7.7)
    mbed-os-tools (0.0.8)

Issue:
When issuing greentea commend to run a test and output results to json, I get the following error.

```
mbedgt --test-spec OUT\test_spec.json -n tests-events-queue* --report-json CY8CKIT_062_WIFI_BT_IAR_results.json

...

mbedgt: exporting to JSON 'CY8CKIT_062_WIFI_BT_IAR_results.json'...
mbedgt: unexpected error:
        write() argument 1 must be unicode, not str
Traceback (most recent call last):
  File "D:\Python\Scripts\mbedgt-script.py", line 11, in <module>
    load_entry_point('mbed-greentea==1.7.1', 'console_scripts', 'mbedgt')()
  File "d:\python\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 358, in main
    cli_ret = main_cli(opts, args)
  File "d:\python\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 1020, in main_cli
    dump_report_to_text_file(opts.report_json_file_name, json_report)
  File "d:\python\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 987, in dump_report_to_text_file
    f.write(content)
TypeError: write() argument 1 must be unicode, not str

```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-11-06:
cc @ARMmbed/mbed-os-test

VeliMattiLahtela, 2019-11-14:
python3 reporting was fixed for mbedgt v1.7.3, could you try with that

0xc0170, 2019-11-25:
Closing as resolved.

closed 2019-11-25 by 0xc0170