ARMmbed/mbed-os #11663: DISCO_H747I - device_name missing in targets.json

JanneKiiskila, 2019-10-10:
### Description

    - What target does this relate to?

DISCO_H747I, 
    - What toolchain (name + version) are you using?

GCC_ARM (irrelevant, though)

    - What tools (name + version - is it mbed-cli, online compiler or IDE) are you using?

mbed cli 1.10.1

    - What is the SHA of Mbed OS (git log -n1 --oneline)?

1798c246cc5c9f66d2c3d37dcb1c6ebb760cff9a

    - Steps to reproduce. (Did you publish code or a test case that exhibits the problem?)

After you add mbed_rom_start, mbed_rom_size to `targets/targets.json` you will hit the next issue - which is the lack of `device_name` -property in the targets.json.

It should likely be 

```
        "device_name": "STM32H747XIH6"
```

But, I would want @ARMmbed/team-st-mcd to confirm.

-->
    [ ] Question
    [ ] Enhancement
    [x] Bug

jeromecoutant, 2019-10-10:
Same as #11617 why ?

jeromecoutant, 2019-10-10:
Sorry for my first wrong comment....

So about device name issue, for me, it's a generic issue in 
https://github.com/ARMmbed/mbed-os/blob/master/tools/arm_pack_manager/index.json

It should be re-generated thanks to latest pack from https://www.keil.com/dd2/pack/

JanneKiiskila, 2019-10-10:
It is actually a tad bit more complicated, we need to update the CMSIS-pack, too. The tooling starts from `targets/targets.json` and then uses the `device_name` value to pick up the data from the CMSIS-pack information in `tools/arm_pack_manager/index.json`. So, we need to update both.

You will easily find out that every target has a `device_name` value simply by doing `git grep device_name targets/targets.json`, including all the earlier STM targets, too.

jeromecoutant, 2019-10-10:
We will add DISCO H747 device_name when #11666 will be corrected.

jeromecoutant, 2019-11-04:
Will be fixed by #11709

closed 2019-11-07 by JanneKiiskila