ARMmbed/mbed-os #11666: Need to update arm_pack_manager file

jeromecoutant, 2019-10-10:
### Description

Several script in MBED are using MBED_RAM_START / MBED_RAM_SIZE / MBED_ROM_START / MBED_ROM_SIZE macro.

These information comes from the pack manager file:
https://github.com/ARMmbed/mbed-os/blob/master/tools/arm_pack_manager/index.json

It is a long time it has not been generated thanks to latest pack from https://www.keil.com/dd2/pack/

Generation script should be even available and documented in order to allow each customer to add the newest chip.

@MarceloSalazar 
@theotherjimmy 

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

jeromecoutant, 2019-10-10:
This generation script has to take into account the new dual core chips.
The extracted "device_name" should be updated to take into account core name.

Ex with STM32H747:
````
        <device Dname="STM32H747XIHx">
          <memory Pname="CM7" name="FLASH_Bank1" access="rx" id="IROM1" start="0x08000000" size="0x00100000" default="1" startup="1"/>
          <memory Pname="CM4" name="FLASH_Bank2" access="rx" id="IROM1" start="0x08100000" size="0x00100000" default="1" startup="1"/>
````

The index.json file should then contain 2 targets:
- STM32H747XIHx_CM7
- STM32H747XIHx_CM4

Thx

@bulislaw 
@ARMmbed/team-st-mcd

JanneKiiskila, 2019-10-10:
@ARMmbed/mbed-os-tools

0xc0170, 2019-10-17:
> Generation script should be even available and documented in order to allow each customer to add the newest chip.

I don't recall how to update but it was updated couple of times with target additions. I'll try to find the notes somewhere.

JanneKiiskila, 2019-10-31:
It is not very straight forward right now, as the cmsis-pack-manager has some TLS handshake issue.

jeromecoutant, 2019-11-07:
I am closing this issue as I don't have any specific issue.
But I think ARM should take about this big pack manager file.

closed 2019-11-07 by jeromecoutant