ARMmbed/mbed-os #11674: STM trng_api uses always RCC_RNGCLKSOURCE_PLL

jarlamsa, 2019-10-11:
### Description

All the STM targets that have TRNG will use `RCC_RNGCLKSOURCE_PLL` even if the
`CLOCK_SOURCE` is defined to be `USE_PLL_MSI` and TRNG should use `RCC_RNGCLKSOURCE_MSI` instead.

This is because the `USE_PLL_MSI` is defined in the `system_clock.c` and isn't seen in the `trng_api.c` thus evaluating to 0 when preprocessing. This causes issues when the same 48MHz clock is used between multiple peripherals and the TRNG modifies the `RCC_CCIPR` -register, while other peripherals think that the MSI-clock is used.

Tested with DISCO_L475VG_IOT01A, seems very likely that all the devices that are using the same STM-specific `trng_api.c` are having similar symptoms.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jarlamsa, 2019-10-11:
ping @MarceloSalazar

jeromecoutant, 2019-10-11:
Hi

Oups... I confirm the issue
Need to add

#define USE_PLL_MSI      0x1 // Use MSI internal clock

jeromecoutant, 2019-10-14:
@0xc0170 could you set it as device st ?
Thx

PR #11679 (2019-10-21): STM32L4 TRNG clock configuration
### Description

Fixes #11674 

In some cases, mainly STM32L4, STM targets that have TRNG will use RCC_RNGCLKSOURCE_PLL even if the CLOCK_SOURCE is defined to be USE_PLL_MSI and TRNG should use RCC_RNGCLKSOURCE_MSI instead.
USE_PLL_MSI was defined in the system_clock.c and isn't seen in the trng_api.c thus evaluating to 0 when preprocessing

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

@jarlamsa 

### Release Notes
closed 2019-10-21 by 0xc0170