ARMmbed/mbed-os #11680: Support entropy source for targets without TRNG H/W

ccli8, 2019-10-14:
### Description

This issue is created for (Nuvoton) targets that don't have TRNG H/W but still want to support mbedtls and Pelion application.

1. For Nuvoton targets like below, they have PRNG H/W but no TRNG H/W. Could come out with an acceptable solution to implementing TRNG HAL with PRNG H/W, or ADC+PRNG? If the criteria requires real TRNG H/W, then PRNG H/W gets of no use. It's a pity.
    - NUMAKER_PFM_NUC472
    - NUMAKER_PFM_M487
    - NUMAKER_IOT_M487

1. Discard TRNG and go NV seed (`MBEDTLS_ENTROPY_NV_SEED`). See that:
    1. For PSA targets, `MBEDTLS_ENTROPY_NV_SEED`, `MBEDTLS_PSA_INJECT_ENTROPY`, `MBEDTLS_PSA_CRYPTO_C`, `MBEDTLS_PSA_CRYPTO_STORAGE_C` are necessary. PSA ITS is implemented with KVStore on secure side.
    1. For non-PSA targets (like NUC472/M487), besides above, `MBEDTLS_PSA_ITS_FILE_C` and `MBEDTLS_FS_IO` are also needed. With `MBEDTLS_PSA_ITS_FILE_C`, PSA ITS is emulated over files, but the files are not protected. With `MBEDTLS_FS_IO`, mbedtls library doesn't compile. I guess on mbed-os, mbedtls with file system access is not supported yet. To go NV seed, could PSA ITS implement with KVStore just like on PSA targets?

#### Related PR

Continuation of https://github.com/ARMmbed/mbed-os/pull/11176#issuecomment-535387417

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

@yanesca @kjbracey-arm @cyliangtw

0xc0170, 2019-10-15:
cc @ARMmbed/mbed-os-crypto

ccli8, 2019-10-17:
> I don't think non-PSA targets need any option other than MBEDTLS_NV_SEED

For mbedtls, enabling `MBEDTLS_ENTROPY_NV_SEED` is enough. But for Pelion, FCC mentions `fcc_entropy_set` for injecting entropy. It will go the sequence: 
1. [fcc_entropy_set](https://github.com/ARMmbed/mbed-cloud-client/blob/84bc0d4582472c07630777da9b8330e2d5ed098a/factory-configurator-client/factory-configurator-client/source/factory_configurator_client.c#L218)
1. [pal_plat_osEntropyInject](https://github.com/ARMmbed/mbed-cloud-client/blob/84bc0d4582472c07630777da9b8330e2d5ed098a/mbed-client-pal/Source/Port/Reference-Impl/Generic/Entropy/PSA/pal_plat_entropy_psa.c#L79)
1. [mbedtls_psa_inject_entropy](https://github.com/ARMmbed/mbed-os/blob/8c860524fb2a130ca2f950d58bcd16e079ef5f3c/features/mbedtls/mbed-crypto/platform/COMPONENT_PSA_SRV_IMPL/psa_crypto.c#L5362)
1. [mbedtls_psa_storage_inject_entropy](https://github.com/ARMmbed/mbed-os/blob/8c860524fb2a130ca2f950d58bcd16e079ef5f3c/features/mbedtls/mbed-crypto/platform/COMPONENT_PSA_SRV_IMPL/psa_crypto_storage.c#L470)

So relevant macros must enable:
- `MBEDTLS_ENTROPY_NV_SEED`
- `MBEDTLS_PSA_INJECT_ENTROPY`
- `MBEDTLS_PSA_CRYPTO_C`
- `MBEDTLS_PSA_CRYPTO_STORAGE_C`
- `MBEDTLS_PSA_ITS_FILE_C`
- `MBEDTLS_FS_IO`

For non-PSA target, per my findings, there are the following issues (repeat above):
1. With `MBEDTLS_PSA_ITS_FILE_C`, `PSA` `ITS` is necessary. But it is implemented over files and not secured.
1. With `MBEDTLS_ENTROPY_NV_SEED`, default implementation of [mbedtls_platform_std_nv_seed_read/mbedtls_platform_std_nv_seed_write](https://github.com/ARMmbed/mbed-os/blob/8c860524fb2a130ca2f950d58bcd16e079ef5f3c/features/mbedtls/mbed-crypto/src/platform.c#L307) is incompatible with above emulated `PSA` `ITS`.
1. With `MBEDTLS_FS_IO`, mbedtls library doesn't compile.

cyliangtw, 2019-10-17:
> 1. Alternative solutions:
>    
>    * HW PRNG: Adding a TRNG HAL integrates it automatically with the entropy collector module of Mbed TLS. The intended way of providing randomness for the library is to seed one of the provided PRNGs (NIST CTR-DRBG and HMAC-DRBG) with entropy provided by the entropy collector module. TRNG is not an alternative to PRNG, it is an input to it. If the hardware PRNG on these targets is cryptographically secure, then applications could potentially use it instead of the software PRNGs provided by Mbed TLS. By the way what algorithm does the HW PRNG implement?
>    * ADC: Using a disconnected ADC as a sole entropy source is arguable at best, and most certainly does not provide equivalent security to a TRNG. Therefore I don't think we can find a solution here that is acceptable as a TRNG for Mbed OS.

@yanesca  In our chip, we have one internal voltage through MCU internal LDO solution. In our measurement, “Even the measured target is a constant voltage output, the ADC transfer result at “low bit” is jumping up and down.  Actually, the low bits result will be impact by the “White Noise”. Even to use the moving average to smooth those raw data, you can also not get a constant or predict value. Though HW TRNG could encapsulate the behavior of seed fetch + PRNG output, in fact, the security of ADC+PRNG is enough for most customers. Moreover, we also could refer to AmazonFreeRTOS platform's mechanism as: _If some use case need the real HW TRNG, application could plug some external TRNG component and override `int mbedtls_hardware_poll( )` by themselves._

yanesca, 2019-10-18:
@ccli8 I think in that case setting  "MBEDTLS_PSA_CRYPTO_C", "MBEDTLS_ENTROPY_NV_SEED" should be enough.

Do you have the lines
> "pal-user-defined-config"                    : "\"pal_config_non_trng_MbedOS.h\"",

and

> "target.extra_labels_add"                    : ["PSA"],

under your target in your `mbed_app.json`? Have you got `"PAL_USE_HW_TRNG=0"` set?

yanesca, 2019-10-18:
@cyliangtw We seem to have different opinion on how secure a floating ADC based entropy source is. However I hope that we can agree that it is less secure than a NIST SP 800-90B compliant TRNG.

There might be a lot of users who think that an ADC based entropy source is sufficient for their purposes, but this needs to be an explicit choice from them. When the user expects a TRNG, because that is what we promise to them in `target.json`, then we shouldn't silently give them an ADC based entropy source instead.

As you mentioned, if a user is happy with the security provided by an ADC based entropy source, then they are free to override `int mbedtls_hardware_poll( )` as they wish.

ccli8, 2019-10-22:
> I think in that case setting "MBEDTLS_PSA_CRYPTO_C", "MBEDTLS_ENTROPY_NV_SEED" should be enough ...

@yanesca Thanks for your information. I've tried it out:

1. Don't define `MBEDTLS_PSA_INJECT_ENTROPY`. Meet `mbedtls_psa_inject_entropy(...)` undefined and then provide custom one.
1. Define `MBEDTLS_ENTROPY_NV_SEED`
1. Define `MBEDTLS_PLATFORM_NV_SEED_READ_MACRO` / `MBEDTLS_PLATFORM_NV_SEED_WRITE_MACRO` and provide custom `mbedtls_nv_seed_read(...)` / `mbedtls_nv_seed_write(...)`.
1. For development, simulate partial provision process by injecting entropy seed via `mbedtls_psa_inject_entropy(...)` pre-main.

Reference configuration in mbed_app.json for Pelion:

```json
"target.macros_add": [
    "MBEDTLS_USER_CONFIG_FILE=\"mbedTLSConfig_mbedOS_SW_TRNG.h\"", 
    "PAL_USE_HW_TRNG=0",
    "MBEDTLS_PLATFORM_NV_SEED_READ_MACRO=mbedtls_platform_seed_read",
    "MBEDTLS_PLATFORM_NV_SEED_WRITE_MACRO=mbedtls_platform_seed_write"
],
```

ccli8, 2019-10-22:
> As you mentioned, if a user is happy with the security provided by an ADC based entropy source, then they are free to override int mbedtls_hardware_poll( ) as they wish.

@yanesca For the custom entropy source via `mbedtls_hardware_poll(...)`, it is workable except for #11725. Thanks for your information.

yanesca, 2019-10-22:
@ccli8 I am happy to hear that it works!

Please be careful and make sure that the functionality matches the specification when implementing `mbedtls_psa_inject_entropy()`:
https://github.com/ARMmbed/mbed-crypto/blob/development/include/psa/crypto_extra.h#L265
For example: it should be callable only one time in the lifetime of the device, every call after the first successful call should return an error.

About  #11725: I left a comment there.

ccli8, 2019-10-23:
> For example: it should be callable only one time in the lifetime of the device, every call after the first successful call should return an error.

@yanesca Thanks for your reminding. It has been taken into consideration in our custom `mbedtls_psa_inject_entropy()`.

Now that we have two alternatives to TRNG: custom entropy source `mbedtls_hardware_poll(...)` and NV seed `mbedtls_psa_inject_entropy()`, I will revoke TRNG support on NUC472/M487 which don't have real TRNG.

0xc0170, 2019-10-23:
@ccli8 Good to hear. Is this issue resolved now?

ccli8, 2019-10-23:
> Is this issue resolved now?

For targets supporting TRNG H/W, #11650 is going.
For targets having no TRNG H/W, after #11725 is merged, we will provide custom entropy source in application example.

closed 2019-10-23 by ccli8