ARMmbed/mbed-os #11722: Issues with SFDP parsing of the Sector Map Parameter Header and Table parsing

miteshdedhia7, 2019-10-21:
### Description
The QSPIFBlockDevice and potentially the SPIFBlockDevice implementation has the following issues,
Commit: fe12608226 (HEAD -> master, origin/master) Merge pull request #11444 from jeromecoutant/PR_QSPI_EXTERNAL
Toolchain: All toolchains
Spec Link: [JESD216](https://www.jedec.org/system/files/docs/JESD216D-01.pdf)

- **BUG**: In QSPIFBlockDevice.cpp, in `_sfdp_parse_sfdp_headers()` function on line 787 `param_header[0] == 81`, the check needs to check for 0x81 (hex) instead of 81 (dec).
Spec Reference: JESD216 pg 54 section 6.5.1

- **BUG/ENHANCEMENT**: The `_sfdp_parse_sector_map_table()` function currently only supports a single map descriptor (no commands) that may not be valid/generic implementation that will work for all flash devices. The spec specifies that there can be variable number of configuration detection commands, configurations and regions.
Spec Reference: JESD216 pg 53 section 6.5

- **BUG**: The  `test_random_program_read_erase` and `test_thread_job` and potentially other tests in features-storage-tests-blockdevice-general_block_device test query the `block_size` using `get_erase_size()`. For a QSPIFBlockDevice, when using the sector map table the 
`get_erase_size()`/`_min_common_erase_size` is calculated by checking if all sectors have a common erase type in `_sfdp_parse_sector_map_table()`. If all sectors in a Flash device  do not have a common erase type (which may be a valid use case) then `get_erase_size()` returns 0 and the the tests will fail since `block_size == 0` with nothing to erase, program or read. Note that `_sfdp_detect_erase_types_inst_and_size` also sets the _min_common_erase_size but this is overriden by _sfdp_parse_sector_map_table() which was never called before due the bug in header parsing mentioned before.

These bugs have been filed in a single ticket since these each of these are related to each other. The can be separated if required.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-10-22:
cc @ARMmbed/mbed-os-storage

ifyall, 2019-11-15:
@maclobdell Cypress will have one board that will be released publicly at the beginning of 2020 that is impacted by this defect. The impact is that we can't use QSPIBlockDevice with this device as the SFDP results for it are not parsed correctly.

Thank you,

Ian

maclobdell, 2019-12-16:
@ARMmbed/mbed-os-storage - Please review and make a suggestion.

VeijoPesonen, 2020-03-02:
@miteshdedhia7 would you please split the bugs as their own issues. The first bug is fixed by the PR #12270

miteshdedhia7, 2020-03-04:
Created issues #12574 #12575 for the other two issues. Updated the information on this ticket with just the header parsing issue.

VeijoPesonen, 2020-04-17:
Fixed by #12270. I'll close this issue.

closed 2020-04-17 by ciarmcom