ARMmbed/mbed-os #11730: Question export mbed website

D252874, 2019-10-22:
### Description

i was looking at your guys there website and found the button for export to desktop ide. When i pressed this button it redirected me to an different page where i can check my board and my ide i am using mcuexpresso it is in that row but it is grey scaled and don't know why?

second in theory if you have an different board and change pins you should be able to flash the program on the mcu or not? because it seems it should work but when i tried that by building via mbed studio the flash is success full but the code is not running.

i saw that it is possible to use the mbed cli to make an version for mcuxpresso but i can't get mbed ide to work this was before i found this

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

D252874, 2019-10-23:
found out why it was gray scaled it is not supported but why ??

0xc0170, 2019-10-23:
This might be better to ask on our forum.

1. Anyway, how to reproduce the first problem with exporters ? What board you have? Often exporters need additional files to produce a valid project. This could be your case, when a board is not yet supported by an exporter (missing definition or template project file).

2. Using different board - yes, it should work. If your app uses Mbed OS provided functionality, there should not be a problem when using different boards.

D252874, 2019-10-23:
> This might be better to ask on our forum.

now you says yes it is probably better to ask on the forum. 

> 1. Anyway, how to reproduce the first problem with exporters ? What board you have? Often exporters need additional files to produce a valid project. This could be your case, when a board is not yet supported by an exporter (missing definition or template project file).

i found out why it was gray scaled becouse i believe for the imxrt1050 evkb you can't use mcuxpresso for some reason even to they both are from nxp probaly something that needs more work

> 2. Using different board - yes, it should work. If your app uses Mbed OS provided functionality, there should not be a problem when using different boards.
yes the program i wrote had mbed os included and it did flash according with no errors but when i tried to run it by resetting the board it doesn't work. wil mbed studio not use the changed pinnames?

0xc0170, 2019-10-24:
If you can't find an error in your application, run a debug session to execute step by step and see what is going on in your app.

> the board is not supported with mbed os.

`MIMXRT1050_EVK` is defined in targets.json file.

I'll close this, if you have further questions, please provide more details and specific questions on our forum : http://forum.mbed.com

closed 2019-10-24 by 0xc0170

D252874, 2019-10-24:
how do you run an debug session when you flashed the code to your mcu

0xc0170, 2019-10-24:
Either you export and flash via IDE (considering will be the same program and the same issue persist, you can then debug). Or if you already have it flashed, attached to running target via gdb (you can use pyOCD for instance).