ARMmbed/mbed-os #11731: Issue with lack of hardware support causing mbed_hal-reset_reason test failure

dustin-crossman, 2019-10-22:
### Description

During the mbed_hal-reset_reason greentea test the device is reset using DAP and the test expects to see a PIN_RESET reason upon reset. The Cypress PSoC6 hardware does not support a reset reason for a DAP reset and is therefore unable to differentiate between a DAP reset and a basic power on. Would it be advisable for the test to additionally accept an UNKNOWN reset reason following the DAP reset? If not, is there a mechanism to skip the test given that it depends on functionality which the hardware does not support?

On a side note, could passing a different reset method to mbedgt (e.g -r pyocd) result in a different reset reason following the DAP reset?

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-10-23:
@ARMmbed/team-cypress Is this known issue - reset reason tests failure?

@dustin-crossman we should not skip a test but rather fix it. There is either missing use case in the test or HAL issue.

closed 2019-10-25 by ciarmcom

dustin-crossman, 2019-10-30:
@0xc0170 @adbridge I posted this over on the forums but on second thought this is probably best filed as a defect. Can this issue be reopened or should I create a new issue here for it?

adbridge, 2019-10-31:
@dustin-crossman We have a new issue template which requires much more detailed information when raising a bug. Perhaps you could separate this original issue into the actual bug section (and raise a new issue) and any questions/enhancements which you could direct to the forum ?