ARMmbed/mbed-os #11732: SPIFBlockDevice doesn't play nice on shared SPI bus

thendrickson, 2019-10-22:
### Description

Target: NRF52840
Toolchain: GCC_ARM
mbed-cli version 1.10.1
mbed-os SHA: b6e5a0a8afa34dec9dae8963778aebce0c82a54b

Flaw is visible by code inspection.
SPIFBlockDevice manipulates it's chip select line without having exclusive access of the spi peripheral. Manipulations of chip select line should all be surrounded by calls to spi.select().

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

kjbracey-arm, 2019-10-23:
Good catch. With current `SPI`, it can deal with the chip select for you - pass `use_gpio_ssel` to the `SPI` constructor, then `_spi.select()` will assert the SSEL, so you can replace all the `_cs = 0` and `_cs = 1` calls with `_spi.select()` and `_spi.deselect()`.

Could you make a patch?

0xc0170, 2019-10-24:
@thendrickson Let us know if you can send a patch. If you do, reference this issue there.  Thanks!

PR #12681 (2020-04-01): SPIFBlockDevice doesn't play nice on shared SPI bus #11732.
Fixes #11732 by use valid constructor SPI

With current `SPI` pass `use_gpio_ssel` to the `SPI` constructor. SPI write command use lock/unlock and select deselect inside. So no more needed to use this methods outside the SPI class.  

### Summary of changes 

#### Impact of changes 

#### Migration actions required
closed 2020-04-01 by 0xc0170