ARMmbed/mbed-os #11733: Issue with greentea results seem to be skewed.

miteshdedhia7, 2019-10-23:
### Description
Attached are two test reports from few greentea runs.

[odd_gt_1.txt](https://github.com/ARMmbed/mbed-os/files/3759260/odd_gt_1.txt)
In this file in the test case report section notice that the tests-mbed_drivers-c_strings suite test shows results for the "deep-sleep - high-speed clocks are turned off", "deep-sleep - source of wake-up - lp ticker" tests which are actually part of the deep sleep tests and not the c_string tests.

[odd_gt_2.txt](https://github.com/ARMmbed/mbed-os/files/3759266/odd_gt_2.txt)
In this file in the test case report section notice that the features-storage-tests-kvstore-static_tests suite shows results for the "Testing callbacks with 0 uchars" test which are part of the call back test suite. 

I am running the following commands
`mbed test --compile -t IAR -m CYW9P62S1_43012EVB_01`
`mbedgt -vV`

Note that I see this issue intermittently. Unfortunately, I only have the result logs and do not have the full logs for two runs above to tell if the test cases actually ran correctly.

Version info:
mbed --version: 1.10.1
mbedgt --version: 1.7.2

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-10-23:
regarding 1st:

Attaching the logs:

```
| CYW9P62S1_43012EVB_01-GCC_ARM | CYW9P62S1_43012EVB_01 | tests-mbed_drivers-c_strings                                                 | deep-sleep - high-speed clocks are turned off                                                     | 0      | 1      | FAIL    | 0.02               |
| CYW9P62S1_43012EVB_01-GCC_ARM | CYW9P62S1_43012EVB_01 | tests-mbed_drivers-c_strings                                                 | deep-sleep - source of wake-up - lp ticker                                                        | 0      | 9      | FAIL    | 0.12               |
| CYW9P62S1_43012EVB_01-GCC_ARM | CYW9P62S1_43012EVB_01 | tests-mbed_drivers-c_strings                                                 | sleep - source of wake-up - us ticker                                                             | 1      | 0      | OK      | 0.02   
```

Not only test case is incorrect but also test results? This looks like a bug in our tooling. Is this target specific?

cc @ARMmbed/team-cypress Have you experienced this?

ifyall, 2019-10-23:
Per @jamesbeyond:
"I think there is some issues with greentea where the test suite name misaligned with test case name, where DUT got reset unexpectedly. 
So in the test log, before print the result summary table, a warning line of “mbedgt: utest test case summary mismatch: utest reported passes and failures miscount!” must have been printed out.
Overall, if the target work just fine passed all test without unexpected reset, the test result should  be just aligned."

@miteshdedhia7 , please check for the "utest case mismatch" warning.

Ian

miteshdedhia7, 2019-10-31:
Yes I have noticed the `utest case mismatch` warning in the logs alert. Unfortunately I do not have the full logs to verify for these specific results as I mentioned earlier in the description. We will look out for that for checking the validity of the test reports.

adbridge, 2020-10-15:
@miteshdedhia7 is this issue still valid ?

miteshdedhia7, 2020-10-15:
@adbridge I dont have a consistent way of reproducing this so I will close this ticket for now and reopen if I run into the problem again.

closed 2020-10-15 by miteshdedhia7