ARMmbed/mbed-os #11737: Need to revert "Enable stats reporting with a flag in targets.json"

VeijoPesonen, 2019-10-23:
Commit 9dbdbe8 - PR #9072 - should be reverted

The reason for reverting the change is that MBED_ROM_START and MBED_ROM_SIZE are needed for configuring bootloader for NRF52840_DK. The change to be reverted is a workaround for an issue with online compiler when using MCU_NRF52832 and MCU_NRF52840 based boards. The workaround prevents generation of those two defines.

The workaround should be removed but the original issue with the online tool should be fixed first, if it's still valid. Please see the original workaround commit #9072 for further details.

High possibility that this breaks online compiler when the forementioned boards are used.

A PR about this have been already created. PR #11510 - Revert "Enable stats reporting with a flag in targets.json" - but it won't get in before it's checked that online compiler continues working with the targets. 

#### Target(s) affected by this defect ?
MCU_NRF52832 and MCU_NRF52840 based devices

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM, ARM, IAR

#### What version of Mbed-os are you using (tag or sha) ?
Mbed OS 5.14.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
Try to compile mbed-bootloader for any target which is based on forementioned SoCs

0xc0170, 2019-10-23:
cc @ARMmbed/mbed-os-tools

loverdeg-ep, 2019-10-25:
@ARMmbed/team-embeddedplanet

trowbridgec, 2019-10-25:
@loverdeg-ep Looks like this is only an issue if the Nordic SoCs are used with SoftDevice (which is now disabled by default), so our nRF52 based boards should be fine in the online compiler at the moment.

adbridge, 2019-11-06:
@VeijoPesonen  you have removed part of the template (ie the Description header) please could you put it back thanks.

VeijoPesonen, 2019-11-06:
@adbridge sorry for that. I hope it was the only one.

loverdeg, 2020-11-22:
@trowbridgec @0xc0170 is this still valid or can it be closed?

adbridge, 2021-06-07:
As there has been no response on this since Nov last year I am closing this. Please re-open if it is still an issue.

closed 2021-06-07 by adbridge