ARMmbed/mbed-os #11740: NetworkInterface connect call hard faults when PSA is enabled on Cypress targets

linlingao, 2019-10-23:
### Description of defect

Pelion example crashes when PSA is enabled in mbed_app.json during NetworkInterface connect().

NetworkInterface connect call triggers init_sdio_bus(), which calls cybsp_get_wifi_sdio_obj(). The sdio_obj retrieved is not initialized. This sdio_obj is initialized in cybsp_init(), but cybsp_init is called only when PSA is disabled, in mbed_overrides.c

#if !defined(TARGET_PSA)
    /* Set up the device based on configurator selections */
    cybsp_init();
    /*
     * Init the us Ticker here to avoid imposing on the limited stack space of the idle thread.
     * This also allows the first call to sleep to occur faster.
     */
    us_ticker_init();
#endif

I'm not sure what the rationale is behind this guard. PSA should have nothing to do with cybsp_init and us_ticker_init.

#### Target(s) affected by this defect ?
CY8CPROTO_062_4343W and possibly other similar targets

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM and possibly all toolchains

#### What version of Mbed-os are you using (tag or sha) ?
5.14.1 and possibly earlier releases

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Shouldn't matter

#### How is this defect reproduced ? 
Build mbed-os-example-pelion for cy8cproto_062_4343w with this PR:
https://github.com/ARMmbed/mbed-os-example-pelion/pull/37

The application should throw an exception.

linlingao, 2019-10-23:
@ARMmbed/team-cypress @maclobdell

maclobdell, 2019-10-23:
Thanks Lin! 

A key aspect of this issue that needs clarification from the security experts (or maybe we need to find where it is documented) - 

What exactly are all the implications, dependencies, requirements, etc for using the TARGET_PSA macro?

In some cases it seems that this macro enables the use of PSA Crypto APIs,
In other cases it seems to indicate that a secure partition manager is enabled.  
In this case for Cypress, it seems they are using the macro to designate that the Cortex M0+ and not the Cortex M4 will handle some of the initialization.

ifyall, 2019-10-31:
@linlingao , I have filed an internal ticket with Cypress to get the extraneous #if !defined(TARGET_PSA) removed.

Thank you,

Ian

morser499, 2019-11-13:
Changes for this are included in https://github.com/ARMmbed/mbed-os/pull/11800

maclobdell, 2019-12-04:
@linlingao - please confirm you are okay to close this issue now.

closed 2019-12-04 by linlingao