ARMmbed/mbed-os #11741: Random FPGA test failures

yarbcy, 2019-10-23:
### Description of defect
**Random FPGA test failures**
Steps to reproduce:
1. Run code 10 or more times:

    // Reset everything and set all tester pins to hi-Z.
    tester.reset();

    // Select GPIO0.
    tester.select_peripheral(MbedTester::PeripheralGPIO);

    // Map pins for test.
    tester.pin_map_set(pin, MbedTester::LogicalPinGPIO0);

    // Initialize GPIO pin with a generic init fun.
    gpio_t gpio;
    gpio_init(&gpio, pin);
    
    // Test GPIO used as an input.
    gpio_dir(&gpio, PIN_INPUT);

    // Test input, pull-up mode.
    gpio_mode(&gpio, PullUp);
    tester.gpio_write(MbedTester::LogicalPinGPIO0, 0, false);
    TEST_ASSERT_EQUAL_INT(1, gpio_read(&gpio)); // hi-Z, pulled up
    
    tester.gpio_write(MbedTester::LogicalPinGPIO0, 0, true);
   TEST_ASSERT_EQUAL_INT(0, gpio_read(&gpio));

Test will fail on second test command. When delete second test command test will PASS.

I captured signals when test failed (test expects logilac 0 but FPGA board set logical 1).

#### Target(s) affected by this defect ?
Tested on CY8CKIT_062_WIFI_BT

#### Toolchain(s) (name and version) displaying this defect ?
Tested on GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
https://github.com/ARMmbed/mbed-os/commit/6240335855e5bac67281c2e122864da53191dd59

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
See steps to reproduce.

mprse, 2019-10-24:
I will try if this also happens on other boards later today or tomorrow.

mprse, 2019-10-24:
I have tested provided code several times on `K64F` and `NUCLEO_F429ZI` running `1000` times in a loop and no error occurred. 
Unfortunately, we don't have `CY8CKIT_062_WIFI_BT` board, but we got `CY8CPROTO_062_4343W`. The problem is that there are no ports on these boards to connect the `FPGA-test-shield`. Also, the pinout is not consistent with Arduino. 
@yarbcy 
Can you share how you connected the `FPGA-test-shield`?
What revision of the `FPGA-test-shield/breakout board` do you have?
Is this failure also exists on `CY8CPROTO_062_4343W`?

yarbcy, 2019-10-24:
Thank. It looks like issue on Cypress side. I already fixed it.

closed 2019-10-24 by yarbcy