ARMmbed/mbed-os #11745: BLE power consumption

Mario-N, 2019-10-24:
### BLE power consumption

In Mbed BLE button example the BLE module IDB05A1 current is too high, average 2.8mA.
I'm measuring the current consumption through Jumper J1 of IDB05A1, so I'm measuring only the current flowing into BLE module. I made some measurement also with oscilloscope and I can see a base current of about 2.8mA with spike during advertising of about 13mA.

#### Nucleo F401RE together with IDB05A1 BLE shield.

#### Mbed Online compiler

#### MBED OS 5.14.0

####  Mbed Online compiler, I tried also MBED studio 0.7.0.16 with release building profiles

#### To reproduce: Nucleo F401R, IDB05A1 BLE shield, Mbed BLE button example

0xc0170, 2019-10-25:
@ARMmbed/team-st-mcd Please review

bsturgess, 2019-10-26:
Hi, I having the same issue on NRF52.

BLE examples (once printf is removed) are resulting in 900ua power consumption.

The issue seems to be related to not being able to deep sleep

Results from logger
Uptime: 2001 Sleep time: 1930 Deep Sleep: 0 Uptime: 4001 Sleep time: 3864 Deep Sleep: 0 Uptime: 6001 Sleep time: 5798 Deep Sleep: 0 Uptime: 8001 Sleep time: 7733 Deep Sleep: 0 Uptime: 10001 Sleep time: 9667 Deep Sleep: 0 Uptime: 12001 Sleep time: 11601 Deep Sleep: 0 Uptime: 14001 Sleep time: 13535 Deep Sleep: 0

jeromecoutant, 2019-10-30:
@avilei

adbridge, 2019-11-06:
@Mario-N You have removed the template fields from the issue header. Could you please put them back and fill in the template accordingly as per the guidelines? Thanks

Mario-N, 2019-11-06:
> @Mario-N You have removed the template fields from the issue header. Could you please put them back and fill in the template accordingly as per the guidelines? Thanks

Thanks Anna,
It should be correct now

Mario-N, 2019-12-17:
Any update regarding this problem? I'm stuck with the project.

40Grit, 2019-12-17:
Can you provide a dump of the peripheral registers right before the WFI/WFE instruction that would put the part to sleep?

Mario-N, 2019-12-17:
I can use MBED Studio, but I need a little instruction on what you exactly need. What is WFI/WFE instruction you mentioned?

avilei, 2019-12-17:
@Mario-N 
if you're measuring via Jumper J1 of X-NUCLEO-IDB05A1, you will get the current consumption of the BlueNRG-MS chip only. The value you read depends on the state the chip, e.g. advertising vs connected, connection parameters, and so on.

Connected mode consumes much less than advertising.
When connected, the longer the connection interval, the lower the consumption.

There is a current estimation tool made by ST that you can use to check the estimated consumption based on different settings. It's available here:
https://www.st.com/content/st_com/en/products/embedded-software/wireless-connectivity-software/stsw-bnrg001.html

Could you please check that the values you get are similar to the ones reported by the tool?

Also, if you need a reference measure with your hardware setup, you may want to try the low-power implementation for BlueNRG-MS based on STM32Cube. The reference for optimized consumption is the SampleApp_DMA_LowPower project in the X-CUBE-BLE1 SW package, available here:
https://www.st.com/content/st_com/en/products/embedded-software/mcu-mpu-embedded-software/stm32-embedded-software/stm32cube-expansion-packages/x-cube-ble1.html

Mario-N, 2019-12-17:
@avilei 
As you said I'm measuring only the current of the BLE module.
I modified the Mbed BLE button example in order to be able to measure the current at various initialization points.
Here the code with the current in mA measured:
`int main()
{
    pc.attach(&onCharReceived);
    while(1)                                                                           3mA
    {
        if(c =='n')
            break;
    }
    c=0;
    pc.printf("Instance\n");
    BLE &ble = BLE::Instance();                                             0.662mA
    while(1)                                  
    {
        if(c =='n')
            break;
    }
    c=0;
    pc.printf("Events\n");
    ble.onEventsToProcess(schedule_ble_events);                0.662mA
    while(1)
    {
        if(c =='n')
            break;
    }
    c=0;
    pc.printf("Demo events\n");
    BatteryDemo demo(ble, event_queue);                          0.662mA
    while(1)
    {
        if(c =='n')
            break;
    }
    c=0;
    pc.printf("Start\n");
    demo.start();                                                                   2.878 - 2.928mA
    pc.printf("Dopo start\n");

    return 0;
}`

Current during Advertising at 1140ms  2.878 - 2.928mA
Current when connected 2.928mA
Measures with Fluke 8060A multimeter.
Compared with the ST tool is really too much.
Then I flashed STM32CubeExpansion SampleApp_SERVER_F401RE.bin. Wow that's the desired current consumption about 2uA during advertising and 33uA when connected.
At this point the HW works great, there must be a bug with cordio stack.

apalmieriGH, 2019-12-17:
@Mario-N 
alternatively, you may use [this pre-cordio implementation](https://github.com/ARMmbed/ble-x-nucleo-idb0xa1) of X-NUCLEO-IDB05A1 driver in order to make another comparison.

Mario-N, 2019-12-17:
@apalmieriGH 
Thanks for suggestion Andrea, do you have an example on how to use that library. I don't know how to use it.

apalmieriGH, 2019-12-17:
@Mario-N 
please, refer to [this one](https://os.mbed.com/teams/ST/code/BLE_HeartRate_IDB0XA1/) for mbed classic.

Mario-N, 2019-12-17:
@apalmieriGH 
Thanks Andrea,
I tryed the example, it works and the consumption is around 2uA. Unfortunately if I use it with MBED-OS 5 it doesn't work. I've already developed all my application using the last OS revision, so this implementation is not feasible for me.
I hope someone could help me.

apalmieriGH, 2019-12-17:
@Mario-N 
[tag mbed-os-5.9.6](https://github.com/ARMmbed/mbed-os-example-ble/tree/mbed-os-5.9.6) of the BLE examples still uses the pre-cordio implementation for BlueNRG.
Hope this can help.

avilei, 2019-12-17:
@Mario-N @MarceloSalazar @pan- 
Andrea's example is an old one using the BLE host implementation of ST's BlueNRG-MS chip.
Latest versions of Mbed OS have embraced the BLE host implementation provided by the Cordio stack.
This raises the question about which is the best course of action when users would like to have alternative BLE host implementations for optimization purposes (e.g reduced footprint and/or low power consumption).

pan-, 2019-12-18:
@avilei is it expected from the BlueNRG-MS chip to consume more power when operating in HCI mode ? Could it be dependent of the version of the firmware being used ?

40Grit, 2019-12-18:
If someone can provide a way to dump the periph registers for both the lower power binary and the latest higher power binary @avilei Can then do a diff on the two dumps and see what Mbed OS is now configuring differently.

This cuts straight to the answer unless silicon errata or other off chip component is involved in the consumption.

@flit anything in pyocd to do this?

Mario-N, 2019-12-18:
As suggested by @apalmieriGH I tested ble with mbed-os-5.9.6 and the consumption is correct, 2uA.
Unfortunately that BLE implementation is not working with the latest m-bed release I have to use with my development due to, vice versa, with 5.9.6 USBMSD is not working.
Any suggestion?

pmancele, 2020-03-18:
Same here, we have a high power consumption with cordio, and we can't send notifications as fast as we used to do previously. Like mentionned in #12602

DBS06, 2020-10-14:
Same here and there #10669

pan-, 2020-10-29:
@avilei @apalmieriGH Is it possible that the sleep management of the `IDB05A1` is not effective when the controller operates in HCI mode instead of ACI mode ? 
The power consumption measured here is the power consumption of the module, not the board.

andy-kogsys, 2021-09-09:
Any updates on this issue? I am facing similar problems on an nRF52.

closed 2021-10-20 by ciarmcom