ARMmbed/mbed-os #11750: hello, who can tell me where is cordio ble stack source code ?

cedar-renjun, 2019-10-26:
hello, who can tell me where is cordio ble stack source code ?

tks

ladislas, 2019-10-26:
I think it's here: https://github.com/ARMmbed/mbed-os/tree/master/features/FEATURE_BLE/targets/TARGET_CORDIO

cederom, 2019-10-27:
+1 :-) 

Cordio website explains source code organization but there is no source code location anywhere :-(
* https://os.mbed.com/docs/mbed-cordio/19.02/stack/folder-organization.html
* https://os.mbed.com/docs/mbed-cordio/19.02/sample-apps/index.html

ladislas, 2019-10-28:
The documentation seems to be out of date, in particular regarding examples which have been move to https://github.com/ARMmbed/mbed-os-example-ble.

The rest is available in the link I shared:

- https://github.com/ARMmbed/mbed-os/tree/master/features/FEATURE_BLE/targets/TARGET_CORDIO/stack
- https://github.com/ARMmbed/mbed-os/tree/master/features/FEATURE_BLE/targets/TARGET_CORDIO

0xc0170, 2019-10-28:
Please use forum.mbed.com for questions.

>The rest is available in the link I shared:

The above comment answers the question. I'll close this as resolved.

closed 2019-10-28 by 0xc0170

cedar-renjun, 2019-10-28:
thanks all, buddy