ARMmbed/mbed-os #11751: Apollo3 Cordio Stack can't enable DLE(payload 251 byte) features with BLE4.2 phone?

cedar-renjun, 2019-10-26:
Recent, we team use apollo3 ble chip( BLE 1M PHY) which have an cordio ip& stack on it

i test apollo3 ble read, write, notify data speed with different phone, then i find

1) apollo3 —> BLE5.0 phone
    packet payload size is 251 bytes, write speed is 15KBps
2) apollo3 —> BLE4.2 phone
    write speed is 3KBps

so, i debug for long time, finally, find 

when apollo3 connect with BLE4.2 phone, BLE stack does not send MTU request RF packet,
even through i have call DmConnSetDataLen function

why and i want enable DLE features to use 251 max payload, when apollo3 connect with BLE4.2 phone
what can i do?

thank you very much, and wish your reply

0xc0170, 2019-10-28:
@cedar-renjun How does this relate to Mbed OS?

cedar-renjun, 2019-10-28:
> @cedar-renjun How does this relate to Mbed OS?

apollo3 use mbed cordio IP core and ble stack

pan-, 2019-10-28:
@cedar-renjun I reiterate @0xc0170 question, does it uses mbed-os ? If so, Cordio APIs shouldn't be called directly. 

Please note that supporting Bluetooth 4.2 doesn't guarantee a larger MTU. When the max MTU size supported is exchanged between both devices, an arbitration is made and the smallest MTU is chosen as the MTU of the connection.  It would be interesting if you provided the phone models.

0xc0170, 2019-11-07:
@cedar-renjun I'll close this as answered. If there are any questions, use our forum http://forum.mbed.com/

closed 2019-11-07 by 0xc0170