ARMmbed/mbed-os #11753: mbed-os cordio ble uart nrf52832

cederom, 2019-10-27:
Hello world :-)

I am switching from Nordic SoftDevice/SDK to ARM Cordio BLE stack. Solution is using BLE UART transport and need one also in Cordio. This is critical requirement. Board is Custom based on `nRF52832` Target.  I am new to Cordio, thus my questions:
* Is BLE UART Profile/Service already supported on Cordio?
* Can I move my Nordic SDK code to Cordio with a simple redefine in `mbed_app.json`?
* Where can I find Cordio BLE UART examples?
* Where can I find Cordio source code repository? (also note https://github.com/ARMmbed/mbed-os/issues/11750).

Using UART I have created a CLI (Command Line Interface) just as on COM/USB_VCOM/TCP that can direct device and get various readings and their history. Can you recommend any better way over some kind of custom data profile that would be better to use than UART? UART does not seem to be preferred in BLE but it is really convenient to use is as Telnet like CLI with easily exapandable funcionalities as opposed to stiff BLE Services/Profiles.

Any hints appreciated :-)

0xc0170, 2019-10-28:
cc @adbridge - two internal tickets created?

@cederom Hi, we are transferring github issue tracker to only contain bugs. The question like this is better suited on forum.mbed.com (there will be blog published soon to explain details).

I'll close this question, discussion should not stop, but definitely post this on forum!

closed 2019-10-28 by 0xc0170

cederom, 2019-10-31:
Dear @adbridge @0xc0170 I would like to reopen this ticket because:
1. There is no dedicated CORDIO repository and this seems to be most sensible place to talk about its _development_.
2. I want to switch to CORDIO but BLE UART is a _mandatory_ requirement.
3. I am willing to help to make BLE UART work on CORDIO with some help from ARM MBED (hints, UUID registration, etc).
4. If there is  no BLE UART in CORDIO until end of year I will stay with Nordic SDK and will have no technical nor business arguments to switch to CORDIO at all.
5. Other people would have similar needs and will also benefit a lot when we create BLE UART for CORDIO.

Thank you for your understanding and I hope for your support, this seems a good motivation to extend CORDIO and create a nice enabled for others as well :-)

0xc0170, 2019-10-31:
> Here is my conversation from MBED Questions and this does not solve my problem: https://os.mbed.com/questions/87391/mbed-os-cordio-ble-uart-nrf52832

We provided more details about the bug tracking https://os.mbed.com/blog/entry/Improving-the-management-of-community-in/

The discussion about Cordio should continue on the forum, I can see Desmond already provided some answers. Let's continue there.

cederom, 2019-10-31:
Okay, will try there, eventually come back when the code is ready for testing.. thanks :-)