ARMmbed/mbed-os #11757: FPGA tests: SPI test cases order issue

yarbcy, 2019-10-28:
### Description of defect

FPGA tests: SPI test case order issue

@0xc0170 @mprse 

I run SPI tests.

Case cases[] = {
    // This will be run for all pins
    Case("SPI - init/free test all pins", all_ports<SPIPort, DefaultFormFactor, spi_test_init_free>),

    // This will be run for all peripherals
    Case("SPI - basic test", all_peripherals<SPIPort, DefaultFormFactor, spi_test_common<SPITester::Mode0, 8, TRANSFER_SPI_MASTER_WRITE_SYNC, FREQ_1_MHZ> >),

Expected is to PASS 2 test cases. First passed, second failed.
When I run them separatelly - each of them passed.

#### Target(s) affected by this defect ?
Tested on CY8CKIT_062_WIFI_BT

#### Toolchain(s) (name and version) displaying this defect ?
Tested on GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
Latest

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/N

#### How is this defect reproduced ? 
Always.

yarbcy, 2019-10-30:
@0xc0170 Could you please take a look?

0xc0170, 2019-10-30:
> @0xc0170 Could you please take a look?

@ARMmbed/mbed-os-hal  to review

mprse, 2019-10-30:
This also looks like a problem with releasing the peripheral. @yarbcy How the test fails? Can you provide the output?

yarbcy, 2019-10-30:
@mprse Looks like the problem is different. Now working on debug. Close this issue.

closed 2019-10-30 by yarbcy