ARMmbed/mbed-os #11760: Gemalto cellular driver fails for EMS31-US modem

jporter4, 2019-10-28:
Gemalto cellular driver fails for EMS31-US modem

### Description of defect

I am working with two similar targets, both based on Nordic nRF52840 and both having a cellular interface.  One target uses a Gemalto EMS31-US cellular modem, the other uses a Gemalto ELS61-US cellular modem.  The two modems have similar AT command sets, but there are some differences.  I discovered one of the differences while tracing the AT commands and responses between the mbed Gemalto cellular driver and the modems on my two targets.

In response to the command "AT^SISS=0,"conId",0<cr>" (Setup Internet Service Profile) from the mbed Gemalto cellular driver:
the ELS61 responds "<cr><ln>OK<cr><ln>", but
the EMS31 responds "<cr><ln>+CME ERROR: 50<cr><ln>". According to the EMS31 AT spec, "conId" must be followed by a number between 1 and 8 inclusive.  Error 50 indicates "Incorrect parameter", as the 0 following "conId" is invalid for this modem (but is valid for the ELS61).

The source of this command is the function "socket_open_defer()" in the file "mbed-os\features\cellular\framework\targets\GEMALTO\CINTERION\GEMALTO_CINTERION_CellularStack.cpp".  The 0 following "conId" is a consequence of the "#define CONNECTION_PROFILE_ID 0" near the top of the file.

Possible fixes:
1. The driver already detects which Gemalto modem it's actually connected to at run-time, so it could use that knowledge to set the "conId" parameter dynamically rather than to a #defined constant.
2. Add "ifndef CONNECTION_PROFILE_ID" / "endif" around the "#define CONNECTION_PROFILE_ID 0".  This would allow CONNECTION_PROFILE_ID to be overridden by EMS31 users in their project files.

#### Target(s) affected by this defect ?
Any target with a Gemalto EMS31-US cellular modem attached.

#### Toolchain(s) (name and version) displaying this defect ?
I'm using IAR EWB for ARM 8.40.  The issue should be observed regardless of which toolchain is used however.

#### What version of Mbed-os are you using (tag or sha) ?
5.14.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
just IAR 8.40

#### How is this defect reproduced ? 
Build for GEMALTO_CINTERION (edit "mbed-os\features\cellular\framework\targets\GEMALTO\CINTERION\mbed_lib.json", set "provide-default" value to true).  Enable AT command tracing to see AT traffic.  Test on a target with ELS61-US cellular modem attached to observe OK response to the "AT^SISS=0,"conId",0<cr>" command.  Test on a target with EMS31-US cellular modem attached to observe "ERROR" response to the same command.

jporter4, 2019-10-28:
ciarmcom - I and a co-worker have both tried clicking multiple times on your link MBOTRIAGE-2285, but it always times out. thanks.

trowbridgec, 2019-10-28:
@jporter4 @ciarmcom is the internal ARM bot, and the link above is for the ARM internal Jira mirror of the public GitHub issues.

jporter4, 2019-10-28:
Hi Chris,
So, the ARM bot added a comment to my issue with a link that I can’t access.  Am I supposed to do something with it ?
Thanks
Jack

Confidentiality Notice: This electronic mail transmission is intended for the use of the individual or entity to which it is addressed and may contain confidential and/or proprietary information belonging to the sender. If you are not the intended recipient, you are hereby notified that any disclosure, use, copying, distribution, or the taking of any action in reliance on the contents of this information is strictly prohibited. If you have received this transmission in error, please notify the sender immediately by e-mail and delete the original message. Thank you for your cooperation.

0xc0170, 2019-10-29:
cc @ARMmbed/mbed-os-wan

AnttiKauppila, 2019-10-29:
@jporter4 Have you tried changing the value of https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/targets/GEMALTO/CINTERION/GEMALTO_CINTERION_CellularStack.cpp#L31 to be between 1 and 8?

jporter4, 2019-10-29:
Yes, if I hard-code it to 1 there it works for the EMS31 but then fails for the ELS61.

AriParkkila, 2019-10-30:
@jporter4 First of all thanks for a very well written bug report :)

`CONNECTION_PROFILE_ID` should probably be replaced with `_cid` to match a connection profile with a PDP context id.

PR #12131 (2019-12-19): Cellular: Fix Gemalto/Cinterion driver

### Summary of changes 

Fix #11760 

Fix #12074 

Added also missing cellular properties in `GEMALTO_CINTERION.cpp`.

#### Impact of changes 

#### Migration actions required
closed 2019-12-19 by adbridge