ARMmbed/mbed-os #11761: Multiple files are including Mbed TLS internal headers

sbutcher-arm, 2019-10-28:
### Description of defect

Multiple source files in Mbed OS are including the Mbed TLS internal file `ssl_internal.h`. This file is supposed to be internal to Mbed TLS and defines APIs which are not part of the external interface of Mbed TLS. There is no guarantee that Mbed TLS will support these interfaces between different versions of Mbed TLS, and code using Mbed TLS should not be using these interfaces and should have no reason to include this header.

The issue can be found by searching for the string `ssl_internal.h`, but specifically is found in the following places:
* [features/device_key/source/DeviceKey.cpp](https://github.com/ARMmbed/mbed-os/blob/acf576a9abcebb47ea67c580ae619c1e514ce915/features/device_key/source/DeviceKey.cpp#L34)
* [features/nanostack/sal-stack-nanostack/source/Security/protocols/tls_sec_prot/tls_sec_prot_lib.c](https://github.com/ARMmbed/mbed-os/blob/917d9829134330219ef6dab66693991265e32248/features/nanostack/sal-stack-nanostack/source/Security/protocols/tls_sec_prot/tls_sec_prot_lib.c#L54)
* [features/lwipstack/lwip/src/apps/altcp_tls/altcp_tls_mbedtls.c](https://github.com/ARMmbed/mbed-os/blob/b0ba3d3dad00163b998d5fa3e2a688c557d9bd91/features/lwipstack/lwip/src/apps/altcp_tls/altcp_tls_mbedtls.c#L82)

As an aside - the fact this mistake has happened in so many places but probably indicates that better documentation is necessary, and that it should be clearer that `ssl_internal.h` should not be included by code using Mbed TLS.

#### Target(s) affected by this defect ?

n/a

#### Toolchain(s) (name and version) displaying this defect ?

n/a

#### What version of Mbed-os are you using (tag or sha) ?

Found in df79609

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

n/a

#### How is this defect reproduced ? 

Source code issue.

0xc0170, 2019-10-29:
cc @ARMmbed/mbed-os-ipcore @ARMmbed/mbed-os-storage

0xc0170, 2019-10-29:
The first two references should be easy to fix as they are our files.

The 3rd one is from upstream repo (lwip) - this should be reported to their tracker to fix it upstream: https://savannah.nongnu.org/bugs/?group=lwip - @ARMmbed/mbed-os-ipcore  is that correct?

michalpasztamobica, 2019-10-29:
After removing the [header from nanostack](https://github.com/ARMmbed/mbed-os/blob/917d9829134330219ef6dab66693991265e32248/features/nanostack/sal-stack-nanostack/source/Security/protocols/tls_sec_prot/tls_sec_prot_lib.c#L54) I could still compile without errors, so it can be removed safely.

For LWIP, however, this may not be so trivial, as it was [included for a reason](https://github.com/ARMmbed/mbed-os/blob/b0ba3d3dad00163b998d5fa3e2a688c557d9bd91/features/lwipstack/lwip/src/apps/altcp_tls/altcp_tls_mbedtls.c#L1052), and the internal functions are actually used. @sbutcher-arm , I assume you'd like that usage removed as well? The code author claims he had issues with expected return values of previous function calls. Is this something you're aware of?