ARMmbed/mbed-os #11769: Cypress: PWM FPGA test wrong assert

yarbcy, 2019-10-29:
### Description of defect
Cypress: PWM FPGA test wrong assert

In file mbed_hal_fpga_ci_test_shield/pwm/main.c line 146 expected value is 10 ms actual (measured) is 10 ms. But log file shows: "1077477377 - peripheral tested on port: peripheral=(1077477377) PWM_OUT=(D7) ...:146::FAIL: Expected 0.100000 Was 10.000000"

I run test case "Case("PWM - period: 10 ms, fill: 10%, api: period/write", one_peripheral<PWMPort, DefaultFormFactor, pwm_period_fill_test<10, 10, PERIOD_WRITE> >),"

#### Target(s) affected by this defect ?
Tested on CY8CKIT_062_WIFI_BT

#### Toolchain(s) (name and version) displaying this defect ?
Tested on GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
Latest

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
Always

yarbcy, 2019-10-30:
@0xc0170 Could you please take a look?

mprse, 2019-10-30:
```
peripheral tested on port: peripheral=(1077477377) PWM_OUT=(D7) ...:146::FAIL: Expected 0.100000 Was 10.000000"
```

According to the requirements:
https://github.com/ARMmbed/mbed-os/blob/419556b84a135c8037bfc162d84074e0ddb38fcc/hal/pwmout_api.h#L42-L43

I the test we are setting output duty-cycle to `10%` (`0.1`). Test fails because `pwmout_read(&pwm_out)` returned `10.0` instead of `0.1`. 

Looks like everything is ok, but the `pwmout_read()` implementation returns the result in the wrong format. The result is represented as a decimal percentage value 10% and we expect 0.1 float value.

https://github.com/ARMmbed/mbed-os/blob/419556b84a135c8037bfc162d84074e0ddb38fcc/targets/TARGET_Cypress/TARGET_PSOC6/cy_pwmout_api.c#L51-L54

`return 100.0f * obj->width_us / obj->period_us;` should be changed to:
`return obj->width_us / obj->period_us;`

yarbcy, 2019-10-30:
@mprse are you going to fix it ?

mprse, 2019-10-30:
I can create a PR, but I don't have a Cypress board to test the fix.

yarbcy, 2019-10-30:
@mprse I can test it on Cypress board.

yarbcy, 2019-10-31:
@mprse Looks good all PWM test PASSED on CY8CKIT_062_WIFI_BT.

yarbcy, 2019-10-31:
[HTST][INF] host test detected: default_auto
[1572537901.11][CONN][INF] found KV pair in stream: {{__testcase_count;37}}, queued...
[1572537901.13][CONN][RXD] >>> Running 37 test cases...
[1572537901.17][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - init/free test}}, queued...
[1572537901.24][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 10%, api: period/write}}, queued...
[1572537901.33][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 10%, api: period_ms/write}}, queued...
[1572537901.40][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 10%, api: period_us/write}}, queued...
[1572537901.49][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 10%, api: period/pulse_width}}, queued...
[1572537901.57][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 10%, api: period/pulse_width_ms}}, queued...
[1572537901.65][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 10%, api: period/pulse_width_us}}, queued...
[1572537901.73][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 50%, api: period/write}}, queued...
[1572537901.80][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 50%, api: period_ms/write}}, queued...
[1572537901.87][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 50%, api: period_us/write}}, queued...
[1572537901.96][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 50%, api: period/pulse_width}}, queued...
[1572537902.04][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 50%, api: period/pulse_width_ms}}, queued...
[1572537902.13][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 50%, api: period/pulse_width_us}}, queued...
[1572537902.20][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 90%, api: period/write}}, queued...
[1572537902.27][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 90%, api: period_ms/write}}, queued...
[1572537902.35][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 90%, api: period_us/write}}, queued...
[1572537902.43][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 90%, api: period/pulse_width}}, queued...
[1572537902.52][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 90%, api: period/pulse_width_ms}}, queued...
[1572537902.59][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 10 ms, fill: 90%, api: period/pulse_width_us}}, queued...
[1572537902.66][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 10%, api: period/write}}, queued...
[1572537902.75][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 10%, api: period_ms/write}}, queued...
[1572537902.82][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 10%, api: period_us/write}}, queued...
[1572537902.90][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 10%, api: period/pulse_width}}, queued...
[1572537902.98][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 10%, api: period/pulse_width_ms}}, queued...
[1572537903.07][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 10%, api: period/pulse_width_us}}, queued...
[1572537903.14][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 50%, api: period/write}}, queued...
[1572537903.22][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 50%, api: period_ms/write}}, queued...
[1572537903.29][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 50%, api: period_us/write}}, queued...
[1572537903.38][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 50%, api: period/pulse_width}}, queued...
[1572537903.46][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 50%, api: period/pulse_width_ms}}, queued...
[1572537903.55][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 50%, api: period/pulse_width_us}}, queued...
[1572537903.62][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 90%, api: period/write}}, queued...
[1572537903.69][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 90%, api: period_ms/write}}, queued...
[1572537903.77][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 90%, api: period_us/write}}, queued...
[1572537903.85][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 90%, api: period/pulse_width}}, queued...
[1572537903.93][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 90%, api: period/pulse_width_ms}}, queued...
[1572537904.02][CONN][RXD] 
[1572537904.02][CONN][INF] found KV pair in stream: {{__testcase_name;PWM - period: 50 ms, fill: 90%, api: period/pulse_width_us}}, queued...
[1572537904.06][CONN][RXD] >>> Running case #1: 'PWM - init/free test'...
[1572537904.11][CONN][INF] found KV pair in stream: {{__testcase_start;PWM - init/free test}}, queued...
[... cut ...]
[1572537935.14][CONN][RXD] ***Testing one PWM pin configuration***
[1572537935.66][CONN][RXD] 1077477377 - peripheral tested on port: peripheral=(1077477377) PWM_OUT=(D7) ...succeeded
[1572537935.75][CONN][INF] found KV pair in stream: {{__testcase_finish;PWM - period: 50 ms, fill: 90%, api: period/pulse_width_us;1;0}}, queued...
[1572537935.84][CONN][RXD] >>> 'PWM - period: 50 ms, fill: 90%, api: period/pulse_width_us': 1 passed, 0 failed
[1572537935.85][CONN][RXD] 
[1572537935.88][CONN][RXD] >>> Test cases: 37 passed, 0 failed

closed 2019-11-13 by yarbcy