ARMmbed/mbed-os #11778: EventQueue: impossible to cancel event with negative unique id

furavikt, 2019-10-30:
### Description of defect
Function 'call_every' from EventQueue class should return unique ID that represents the posted event or an ID of 0 if there is not enough memory. If ID is greater than 0, everything is perfect. Sometimes it returns negative value (different) and such event is not working properly, ie event is running well but function 'time_left' returns invalid value -1 and it is not possible to cancel this event (it's running forever). 

#### Target(s) affected by this defect ?
NUCLEO-F439ZI 

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.13.4 and mbed-os-5.10.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
Run this code. If is everything perfect, it will prints everytimes 10 numbers  per a line to console. Everytimes the event with negative ID occurs, the amount of number increases by 10.

```
Serial      pc(SERIAL_TX, SERIAL_RX); // tx, rx
Thread      t;
EventQueue  _queue;
int16_t     id, cntr = 0;

void dummy_function(){
    pc.printf("%02d ", cntr);
}

void main(){
    pc.printf("Dispatch event queue from threads context\n");
    t.start(callback(&_queue, &EventQueue::dispatch_forever));

    while(1){
        // Create new event
        id = _queue.call_every(100, callback(&dummy_function));
        if(id == 0){
            pc.printf("ERROR\n\n");
        } else {
            pc.printf("\nCreated periodic event with ID %d (time_left %d)\n", id, _queue.time_left(id));
        }
        ThisThread::sleep_for(1000);

        // Cancel event
        pc.printf("\nCancel event with ID %d (time_left %d)\n", id, _queue.time_left(id));
        _queue.cancel(id);
        ThisThread::sleep_for(100);

        cntr++;
    }
}
```

0xc0170, 2019-10-31:
cc @ARMmbed/mbed-os-core

kjbracey-arm, 2019-10-31:
I've looked at this, and I can see potential flaws, but I'm not sure how you're getting a negative ID in your example.

The ID you're given is formed from an 8-bit internal ID shifted up above a buffer offset, which is sized according to the size of the buffer.

For the combined result to be negative, the event queue buffer size would have to be 24-bit (so >= 8MiB), as far as I can see.

I'll post a patch making some changes, and we can see if it helps anything. But my test requires a big buffer...?

kjbracey-arm, 2019-10-31:
Ah, running your specific example, I note that you're putting the ID into a `int16_t`. This issue is that that assignment is overflowing. With the variable being `int`, there should be no problem (until the queue size is >=8MiB).

So I think this is "not a bug", but I'll still make the other change, just in case.

furavikt, 2019-11-01:
@kjbracey-arm Good point, the ID in example above is only 16 bit number and that's the problem. In my real code there was also a 16 bit temporal variable for debug and it overflows... I apologize for the stupid mistake. The EventQueue is working properly until the queue size is >= 8MiB. Thank you for your time.

PR #11782 (2019-11-06): equeue: avoid non-standard bit shifts
### Description (*required*)

Shifting negative numbers right is implementation-defined, and shifting positive signed numbers left and exceeding positive range is undefined.

Take care to make sure we always shift unsigned values.

Attempt to fix #11778 

----------------------------------------------------------------------------------------------------------------
### Pull request type (*required*)

    [X] Patch update (Bug fix / Target update / Docs update / Test update / Refactor)
    [] Feature update (New feature / Functionality change / New API)
   [... cut]
closed 2019-11-01 by furavikt