ARMmbed/mbed-os #11787: Dependency on Microsoft Visual C++ Build Tools

MarceloSalazar, 2019-10-31:
### Description of defect

While installing Mbed OS 5.14 from scratch, and the `requirements.txt`, the tools complain about not being able to find Microsoft Visual C++ Build Tools.
However, it **should not** be a requirement to have 3rd party tools such as Microsoft Visual C++ Build Tools.

```
Collecting pycryptodome<=3.7.3,>=3.7.2 (from -r requirements.txt (line 22))
  Using cached https://files.pythonhosted.org/packages/c7/ff/1ca71a40eb69c89778396a30d399639d41473b09c36aff2b700d80dd94b9/pycryptodome-3.7.3.tar.gz
    ERROR: Command errored out with exit status 1:
     command: 'c:\users\maslad\appdata\local\programs\python\python38-32\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\maslad\\AppData\\Local\\Temp\\pip-install-1lr02x0k\\pycryptodome\\setup.py'"'"'; __file__='"'"'C:\\Users\\maslad\\AppData\\Local\\Temp\\pip-install-1lr02x0k\\pycryptodome\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base pip-egg-info
         cwd: C:\Users\maslad\AppData\Local\Temp\pip-install-1lr02x0k\pycryptodome\
    Complete output (31 lines):
    Traceback (most recent call last):
      File "c:\users\maslad\appdata\local\programs\python\python38-32\lib\site-packages\setuptools\msvc.py", line 489, in _find_latest_available_vc_ver
        return self.find_available_vc_vers()[-1]
    IndexError: list index out of range

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\maslad\AppData\Local\Temp\pip-install-1lr02x0k\pycryptodome\setup.py", line 445, in <module>
        set_compiler_options(package_root, ext_modules)
      File "C:\Users\maslad\AppData\Local\Temp\pip-install-1lr02x0k\pycryptodome\compiler_opt.py", line 303, in set_compiler_options
        clang = compiler_is_clang()
      File "C:\Users\maslad\AppData\Local\Temp\pip-install-1lr02x0k\pycryptodome\compiler_opt.py", line 238, in compiler_is_clang
        return test_compilation(source, msg="clang")
      File "C:\Users\maslad\AppData\Local\Temp\pip-install-1lr02x0k\pycryptodome\compiler_opt.py", line 81, in test_compilation
        objects = compiler.compile([fname], extra_postargs=extra_cc_options)
      File "c:\users\maslad\appdata\local\programs\python\python38-32\lib\distutils\_msvccompiler.py", line 360, in compile
        self.initialize()
      File "c:\users\maslad\appdata\local\programs\python\python38-32\lib\distutils\_msvccompiler.py", line 253, in initialize
        vc_env = _get_vc_env(plat_spec)
      File "c:\users\maslad\appdata\local\programs\python\python38-32\lib\site-packages\setuptools\msvc.py", line 185, in msvc14_get_vc_env
        return EnvironmentInfo(plat_spec, vc_min_ver=14.0).return_env()
      File "c:\users\maslad\appdata\local\programs\python\python38-32\lib\site-packages\setuptools\msvc.py", line 843, in __init__
        self.si = SystemInfo(self.ri, vc_ver)
      File "c:\users\maslad\appdata\local\programs\python\python38-32\lib\site-packages\setuptools\msvc.py", line 485, in __init__
        self.vc_ver = vc_ver or self._find_latest_available_vc_ver()
      File "c:\users\maslad\appdata\local\programs\python\python38-32\lib\site-packages\setuptools\msvc.py", line 492, in _find_latest_available_vc_ver
        raise distutils.errors.DistutilsPlatformError(err)
    distutils.errors.DistutilsPlatformError: Microsoft Visual C++ 14.0 is required. Get it with "Microsoft Visual C++ Build Tools": https://visualstudio.microsoft.com/downloads/
    Testing support for clang
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

#### Target(s) affected by this defect ?
Any

#### Toolchain(s) (name and version) displaying this defect ?
Any

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os (#b6e5a0a8afa3, tags: latest, mbed-os-5.14.0)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
1.10.1

#### How is this defect reproduced ? 
 pip install -r requirements.txt

0xc0170, 2019-10-31:
This is unfortunate.

This module came in with tfm update to their tooling. @devran01 Can you please review, is this module still required? Is there a way not to include the compiler?

devran01, 2019-10-31:
@MarceloSalazar @0xc0170 Yes, it was added as a dependency because of image signing scripts of tf-m and they are still needed. `pycryptodome` supports wheel and it should be automatically picked up by pip. I'm surprised it is not.
@MarceloSalazar  what is the version of pip3?

MarceloSalazar, 2019-11-05:
```
python --version
Python 3.8.0

pip --version
pip 19.3.1 from c:\users\...\appdata\local\programs\python\python38-32\lib\site-packages\pip (python 3.8)
```

MarceloSalazar, 2019-11-07:
Find here an update:

Installed wheel:
```
$ pip install wheel
Collecting wheel
  Downloading https://files.pythonhosted.org/packages/00/83/b4a77d044e78ad1a45610eb88f745be2fd2c6d658f9798a15e384b7d57c9/wheel-0.33.6-py2.py3-none-any.whl
Installing collected packages: wheel
Successfully installed wheel-0.33.6
```

Try again:
```
$ mbed compile -t ARM -m NUCLEO_F429ZI
...
ModuleNotFoundError: No module named 'elftools'
```

Install elftools
```
$ pip install elftools
...
 error: Microsoft Visual C++ 14.0 is required. Get it with "Microsoft Visual C++ Build Tools": https://visualstudio.microsoft.com/downloads/
```

@ARMmbed/mbed-os-tools who can help to investigate and fix this?

mark-edgeworth, 2019-11-11:
Works for me using blinky as an example. I do not have (or need) Microsoft tooling.

mark-edgeworth, 2019-11-11:
Now using program mbed-cloud-client-example, and following this with a `mbed dm init -d "arm.com" --model-name "mbed" --force -q`

Still compiles ok on a mac.
Grep through mbed-os and there are references to elftools, but requirements file installs 'pyelftools'

MarceloSalazar, 2019-11-11:
Discussed with @mark-edgeworth and this continues to be a problem on Windows. It seems this dependency is resolved on Mac somehow.
@devran01 we need to find the way to remove/replace `pycryptodome`. Lets discuss this when are around.

devran01, 2019-11-14:
@MarceloSalazar @mark-edgeworth created https://github.com/ARMmbed/mbed-os/pull/11866

closed 2019-12-16 by 0xc0170