ARMmbed/mbed-os #11793: mbed-bootloader wit SPI flash storage hangs at startup

toyowata, 2019-11-01:
### Description of defect

mbed-bootlader with SPI flash storage hangs at start up.  I didn't see the issue on Mbed OS 5.13.x

#### Target(s) affected by this defect ?

Targets which has external SPI flash storage.

#### Toolchain(s) (name and version) displaying this defect ?

Any (GCC_ARM, ARM and IAR)

#### What version of Mbed-os are you using (tag or sha) ?

```
[mbed] Working path "D:\work\mbed-bootloader" (program)
mbed-bootloader (#663a861a3d08, tag: v4.1.0)
`- mbed-os (#f60e9a8075ea, tag: mbed-os-5.14.0-rc2)
```

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

```
$ mbed --version
1.10.1
```

#### How is this defect reproduced ? 

* Import mbed-bootloader

```shell
$ mbed import mbed-bootloader
$ cd mbed-bootloader
```

* Add target config to `configs\internal_kvstore_with_spif.json` file as below (I needed it because default build made link error for lacking of 32kB memory region)

```json
        "DISCO_L475VG_IOT01A": {
            "target.restrict_size"           : "0x9000",
            "mbed-bootloader.bootloader-size": "(36*1024)"
        }
```

* Build the binary

```shell
$  mbed compile -m DISCO_L475VG_IOT01A -t gcc_arm --profile .\tiny.json --app-config .\configs\internal_kvstore_with_spif.json
```
* Ran program and I got log below (it just hangs):

```
Mbed Bootloader
```

#### Possible workaround

The mbed-bootloader has `bare-metal` profile and it hangs at SPI flash storage initialization here:

https://github.com/ARMmbed/mbed-os/blob/c93c181b6a0c3f08a9aab783a81c3152cdb9d79f/components/storage/blockdevice/COMPONENT_SPIF/SPIFBlockDevice.cpp#L958

I locally made the patch below and it works fine (no hangs and bootloader jumps to app location).

```cpp
#if MBED_CONF_RTOS_PRESENT
        rtos::ThisThread::sleep_for(1);
#endif
```

Boot log:
```
Mbed Bootloader
booting...

Starting Simple Pelion Device Management Client example
```

JanneKiiskila, 2019-11-07:
Ping @ARMmbed/mbed-os-storage

VeijoPesonen, 2019-11-07:
@toyowata would you please try removing [this](https://github.com/ARMmbed/mbed-bootloader/blob/663a861a3d0869d8d2cdabded5049bf9f83b8a80/configs/internal_kvstore_with_spif.json#L98) line. @JammuKekkonen was having the same issue with different ST board. I assume the fix he created should work also in this case. I'll create a PR but checking it beforehand would be nice.

VeijoPesonen, 2019-11-07:
@toyowata If you're using the external flash on the board you should rely on the QSPIF driver.

toyowata, 2019-11-11:
@VeijoPesonen Thank you for your response.

> would you please try removing this line. 

Removing the line (means enabling LPTICKER feature) fixes the issue.  Thanks.

VeijoPesonen, 2019-11-21:
I assume this issue can be closed then.

closed 2019-11-22 by 0xc0170