ARMmbed/mbed-os #11795: Cypress: SPI FPGA test: tester always respond 0 when MODE other then 0

yarbcy, 2019-11-01:
### Description of defect
Cypress: SPI FPGA test: tester always respond 0 when MODE other then 0.

#### Target(s) affected by this defect ?
Tested on CY8CKIT_062_WIFI_BT

#### Toolchain(s) (name and version) displaying this defect ?
Tested on GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
Latest
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
Always.

yarbcy, 2019-11-01:
@0xc0170 @mprse Please take a look.

mprse, 2019-11-04:
It's hard to guess what is wrong without the board. The following additional information would be helpful:
 - test output,
 - does the test pass if only single `SPI - mode testing (MODE_1)` test case is executed?
 - can you decrease number of transferred symbols in the test to 5 (`TRANSFER_COUNT = 5`) and provide a view from the logic analyzer for  `SPI - mode testing (MODE_1)` test case?

yarbcy, 2019-11-04:
@mprse 
**1) Test output:** 

> 080164352 - peripheral tested on port: peripheral=(1080164352) MOSI=(A0) MISO=(A1) SCLK=(A2) SSEL=(A3) ...:109::FAIL: Expected 1 Was 0
[1572858689.45][CONN][INF] found KV pair in stream: {{__testcase_finish;SPI - basic test;0;1}}, queued...
[1572858689.53][CONN][RXD] >>> 'SPI - basic test': 0 passed, 1 failed with reason 'Assertion Failed'
[1572858689.53][CONN][RXD] 
[1572858689.60][CONN][RXD] >>> Test cases: 0 passed, 1 failed with reason 'Assertion Failed'
[1572858689.61][CONN][RXD] >>> TESTS FAILED!

> 
2) No it doesn't.
3) It looks like no sense for me because when I set (TRANSFER_COUNT = 5) even MODE0 FAILED (PASSED when (TRANSFER_COUNT = 300)).

mprse, 2019-11-04:
Looking at the provided log. It applies to `SPI - basic test` test case (this test case uses `MODE == 0`). It fails because unexpected data is received from SPI-slave (FPGA-Test-Shield). 

In the test scenario we expect to have `TRANSFER_COUNT` 8-bit symbols transmitted in both directions:
```
MOSI: 0x00 0xFF 0xFE 0xFD ...
MISO: 0x00 0x01 0x02 0x03 ...
```

I suggest checking what is transmitted on the logic analyzer. 
You can also try to comment out line `109` (allow the test to move forward) and verify the number of received symbols/checksum on the SPI-slave side (FPGA-Test-Shield) - lines 157, 158.

yarbcy, 2019-11-04:
@mprse Please notice that log file is related to  MODE == 1. It because I manualy change MODE == 0 -> MODE == 1 in line 175. Test case with MODE==0 passed witout any problems.

yarbcy, 2019-11-04:
@mprse 
**Mode==0 log:**

> 1080164352 - peripheral tested on port: peripheral=(1080164352) MOSI=(A0) MISO=(A1) SCLK=(A2) SSEL=(A3) ...succeeded
[1572863657.66][CONN][RXD] 1080492032 - Could not find pins to test peripheral
[1572863657.71][CONN][INF] found KV pair in stream: {{__testcase_finish;SPI - basic test;1;0}}, queued...
[1572863657.76][CONN][RXD] >>> 'SPI - basic test': 1 passed, 0 failed
[1572863657.76][CONN][RXD] 
[1572863657.79][CONN][RXD] >>> Test cases: 1 passed, 0 failed

> 

**Mode==1 log:**

> 1080164352 - peripheral tested on port: peripheral=(1080164352) MOSI=(A0) MISO=(A1) SCLK=(A2) SSEL=(A3) ...:109::FAIL: Expected 1 Was 0
[1572863733.97][CONN][INF] found KV pair in stream: {{__testcase_finish;SPI - mode testing (MODE_1);0;1}}, queued...
[1572863734.07][CONN][RXD] >>> 'SPI - mode testing (MODE_1)': 0 passed, 1 failed with reason 'Assertion Failed'
[1572863734.07][CONN][RXD] 
[1572863734.13][CONN][RXD] >>> Test cases: 0 passed, 1 failed with reason 'Assertion Failed'
[1572863734.15][CONN][RXD] >>> TESTS FAILED!

>

mprse, 2019-11-04:
OK. 
I strongly suggest connecting the logic analyzer. This will prove that signals states are valid in `MODE 1` (`sclk` idle low/sampling on the second edge).

yarbcy, 2019-11-04:
Please see attachment.

yarbcy, 2019-11-04:
This is wrong attachment.

пн, 4 лист. 2019 о 13:35 Yaremkiv Bogdan <yaremkivb@gmail.com> пише:

mprse, 2019-11-04:
I think you can drag the image file and drop it in the text box.

yarbcy, 2019-11-04:
[image]

mprse, 2019-11-04:
In `MODE 1` `sclk` idle state is low and on the logic analyzer view, we can see that `sclk` idle state is high.

https://github.com/ARMmbed/fpga-ci-test-shield/blob/f69c6c393ecbd20fd80fadf2a7ed48b49457a905/rtl/spi_slave.v#L60

Please try to switch `CYHAL_SPI_MODE_01_MSB` to `CYHAL_SPI_MODE_10_MSB`:

https://github.com/ARMmbed/mbed-os/blob/f59d71f0b40bedcd0181012f86ff5cfe6540d94e/targets/TARGET_Cypress/TARGET_PSOC6/cy_spi_api.c#L98-L112

mprse, 2019-11-04:
`spi_format()` calls `cyhal_spi_init()` with `hal_mode == CYHAL_SPI_MODE_01_MSB == 2`:

https://github.com/ARMmbed/mbed-os/blob/f59d71f0b40bedcd0181012f86ff5cfe6540d94e/targets/TARGET_Cypress/TARGET_PSOC6/cy_spi_api.c#L98-L115

`cyhal_spi_init()` calls `cyhal_convert_mode_sclk()` with `mode == CYHAL_SPI_MODE_01_MSB == 2`:
https://github.com/ARMmbed/mbed-os/blob/f59d71f0b40bedcd0181012f86ff5cfe6540d94e/targets/TARGET_Cypress/TARGET_PSOC6/psoc6csp/hal/src/cyhal_spi.c#L394-L395

I believe that `cyhal_convert_mode_sclk()` returns `CY_SCB_SPI_CPHA0_CPOL1` in our case. That is why we have `sclk` idle state high:

https://github.com/ARMmbed/mbed-os/blob/f59d71f0b40bedcd0181012f86ff5cfe6540d94e/targets/TARGET_Cypress/TARGET_PSOC6/psoc6csp/hal/src/cyhal_spi.c#L163-L180

mprse, 2019-11-04:
I guess that the following comments are invalid:

https://github.com/ARMmbed/mbed-os/blob/f59d71f0b40bedcd0181012f86ff5cfe6540d94e/targets/TARGET_Cypress/TARGET_PSOC6/psoc6csp/hal/include/cyhal_spi.h#L86-L87

Should be:
`/** Standard motorola SPI CPHA=0, CPOL=1 with MSB first operation */`

To be consistent with:
https://github.com/ARMmbed/mbed-os/blob/f59d71f0b40bedcd0181012f86ff5cfe6540d94e/targets/TARGET_Cypress/TARGET_PSOC6/psoc6pdl/drivers/include/cy_scb_spi.h#L361

mprse, 2019-11-04:
@yarbcy Any updates?

yarbcy, 2019-11-04:
[image]

yarbcy, 2019-11-04:
@mprse I made debug session while calling SPI_Init().

yarbcy, 2019-11-04:
@mprse  Is it what you expected? (CY_SCB_SPI_CPHA0_CPOL1)

mprse, 2019-11-04:
I see we expect `CPHA_1_CPOL_0` form `MODE 1`.

yarbcy, 2019-11-04:
Test case PASSED with MODE=1 and CY_SCB_SPI_CPHA1_CPOL0

yarbcy, 2019-11-04:
Created internal merge request.

mprse, 2019-11-05:
Are you going to create a fix for Cypress SPI driver in `Mbed-os`?

yarbcy, 2019-11-05:
Yes.

yarbcy, 2019-11-05:
Yes.

0xc0170, 2019-11-05:
Nice work !

yarbcy, 2019-11-05:
#11813

closed 2019-11-14 by yarbcy