ARMmbed/mbed-os #11798: STM32F207ZGTx in tools/arm_pack_manager/index.json not populated correctly

JanneKiiskila, 2019-11-01:
### Description of defect

See originating PR; https://github.com/ARMmbed/mbed-os/pull/11707

The Keil CMSIS-pack file has this for the

```
        <!-- *************************  Device 'STM32F207ZG'  ***************************** -->
        <device Dname="STM32F207ZG">
          <memory id="IROM1"                               start="0x08000000" size="0x00100000" startup="1" default="1"/>
          <memory id="IRAM1"                               start="0x20000000" size="0x00020000" init   ="0" default="1"/>
          <algorithm name="CMSIS/Flash/STM32F2xx_1024.FLM" start="0x08000000" size="0x00100000"             default="1"/>
          <algorithm name="CMSIS/Flash/STM32F2xx_OPT.FLM"  start="0x1FFFC000" size="0x00000010"             default="0"/>
          <algorithm name="CMSIS/Flash/STM32F2xx_OTP.FLM"  start="0x1FFF7800" size="0x00000210"             default="0"/>
          <feature type="ADC"           n="24"      m="12"/>
          <feature type="IOs"           n="114"/>
          <variant Dvariant="STM32F207ZGTx"> <feature type="QFP" n="144"/> </variant>
        </device>
```

The current `index.json` under `tools/arm_pack_manager` has this information:

```
        "name": "STM32F207ZGTx", 
        "processor": {
            "Symmetric": {
                "core": "CortexM3", 
                "fpu": "None", 
                "mpu": "Present", 
                "units": 1
            }
        }, 
        "sectors": [
            [
                134217728, 
                16384
            ], 
            [
                134283264, 
                65536
            ], 
            [
                134348800, 
                131072
            ], 
            [
                536836096, 
                528
            ], 
            [
                536854528, 
                16
            ]
        ], 
```

So, the information seems to be essentially there, but it is not parsed out somehow / generated to the output.

Output is:

```
    "name": "STM32F207ZGTx", 
    "processor": {
        "Symmetric": {
            "core": "CortexM3", 
            "fpu": "None", 
            "mpu": "Present", 
            "units": 1
        }
    }, 
    "sectors": null, 
```

@theotherjimmy analyzed root cause to be that CPM wrapper in Mbed OS is not populating the fields correctly (see https://github.com/ARMmbed/cmsis-pack-manager/issues/126).

#### Target(s) affected by this defect ?

STM32F207ZGTx

#### Toolchain(s) (name and version) displaying this defect ?

mbed cli v1.10.1

#### What version of Mbed-os are you using (tag or sha) ?

5.14.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

Clone some repo, which pulls in mbed-os, too.

```
cd mbed-os
python project.py -m STM32F7xx --update-packs
```
analyze the produced `index.json` file in `tools/arm_pack_manager`.

JanneKiiskila, 2019-11-01:
@ARMmbed/mbed-os-tools

mark-edgeworth, 2019-11-04:
What should the sectors entry look like for the example above?

JanneKiiskila, 2019-11-04:
The ones we have in the currently merged `index.json` in Mbed OS master (and previous release branches) has it correctly.

JanneKiiskila, 2019-12-22:
We have now the same problems also with STM32WB and STM32L4 series.

jeromecoutant, 2020-01-02:
We got now 4 referenced issues waiting for this one...

@ARMmbed/mbed-os-tools please have a look

@MarceloSalazar 
@bulislaw

bulislaw, 2020-01-02:
@jeromecoutant it was escalated to a blocker internally, @ARMmbed/mbed-os-tools should look at it asap.

0xc0170, 2020-01-10:
> @jeromecoutant it was escalated to a blocker internally, @ARMmbed/mbed-os-tools should look at it asap.

Any update with this one?

jeromecoutant, 2020-01-20:
@bulislaw 

> it was escalated to a blocker internally, @ARMmbed/mbed-os-tools should look at it asap.

Any update?

JanneKiiskila, 2020-01-20:
Seems to me that most of CMSIS-pack files do not even have the required information to populate it necessarily (at least if the sector info should be the flash erase sector size information).

bulislaw, 2020-01-20:
@ARMmbed/mbed-os-tools any updates on this?