ARMmbed/mbed-os #11799: GreenTea: can not create txt test report

yarbcy, 2019-11-01:
### Description of defect
GreenTea: can not create txt test report

Steps to reproduce:
1. mbed import mbed-os
2. cd mbed-os
3. mbed test --compile -m CY8CKIT_062_BLE -t GCC_ARM -n TESTS-mbed_hal-gpio
4. mbed test --run -m CY8CKIT_062_BLE -t GCC_ARM -n TESTS-mbed_hal-gpio --report-text=greentea.txt

Expected:
Test report created without errors

Actual:

> mbedgt: exporting to TEXT 'greentea.txt'...
mbedgt: unexpected error:
        'str' object has no attribute 'decode'
Traceback (most recent call last):
  File "/usr/local/bin/mbedgt", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/mbed_greentea/mbed_greentea_cli.py", line 358, in main
    cli_ret = main_cli(opts, args)
  File "/usr/local/lib/python3.7/dist-packages/mbed_greentea/mbed_greentea_cli.py", line 1012, in main_cli
    dump_report_to_text_file(opts.report_text_file_name, text_final_report)
  File "/usr/local/lib/python3.7/dist-packages/mbed_greentea/mbed_greentea_cli.py", line 986, in dump_report_to_text_file
    f.write(content.decode('utf-8'))
AttributeError: 'str' object has no attribute 'decode'
[mbed] ERROR: "mbedgt" returned error.
       Code: 1
       Path: "/home/pi/bsp_csp/mbed/mbed-os"
       Command: "mbedgt --test-spec ./BUILD/tests/CY8CKIT_062_BLE/GCC_ARM/test_spec.json -n TESTS-mbed_hal-gpio --report-text=greentea.txt"
       Tip: You could retry the last command with "-v" flag for verbose 

> 

Reproduced on RaspberryPi:

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

#### Target(s) affected by this defect ?
All

#### Toolchain(s) (name and version) displaying this defect ?
All

#### What version of Mbed-os are you using (tag or sha) ?
https://github.com/ARMmbed/mbed-os
https://github.com/ARMmbed/mbed-os/commit/c93c181b6a0c3f08a9aab783a81c3152cdb9d79f
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
Always

dustin-crossman, 2019-11-01:
So talking with @yarbcy internally I believe this is because our test machines currently have mbed-greentea version 1.7.2 as installed by pip which have the f.write(content.decode() call that is throwing the exception. When attempting to upgrade them to 1.7.3 (which should have the decode call removed: https://github.com/ARMmbed/mbed-os-tools/commit/f231376f85dbc4c253fce471a8df911d0e4c88ad#diff-6efaa6d22649fc7ca40376a1bbf505d4) I was unable to with pip saying there was no version 1.7.3.

It appears the reason I'm unable to upgrade is that mbed-greentea version 1.7.3 on pypi is tagged as python version "py2": https://pypi.org/project/mbed-greentea/1.7.3/#files whereas older versions are tagged as python version "None": https://pypi.org/project/mbed-greentea/1.7.2/#files

Is 1.7.3 intended to be python 2 only?

0xc0170, 2019-11-04:
cc @ARMmbed/mbed-os-tools Please review

dustin-crossman, 2019-11-08:
Is there any update on this?

0xc0170, 2019-11-08:
If I run locally: `pip install -U mbed-greentea` , I only get 1.7.2 if I execute this for python 3.7.3.

Previous releases contain None, I'll talk to tools team to push updated version. I can't find the change in the repository itself so had  to be published differently.

cc @maclobdell

mark-edgeworth, 2019-11-13:
greentea is not covered by @ARMmbed/mbed-os-tools; @jamesbeyond may be interested...

yarbcy, 2019-11-13:
@0xc0170 Please check. We updated greentea and issue is still reproducuble:
Name: mbed-greentea
Version: 1.7.3

jamesbeyond, 2019-11-13:
Seem this is PypI release problem, I can reproduce what @0xc0170  got:

with python2 .7, I can install greentea 1.7.3 by `pip install mbed-grentea==1.7.3`
but with python3.x , greentea 1.7.3 is not available, the latest is 1.72.
any ideas @mark-edgeworth, I don't control of PyPI side

jamesbeyond, 2019-11-13:
BTW @yarbcy , on my environment  of python2.7,  the greentea 1.7.3 works just fine with text report generate. please confirm your python version? I assume it is 3.x ?

yarbcy, 2019-11-13:
@jamesbeyond  3.x

jamesbeyond, 2019-11-13:
on my environment I could not install greentea 1.7.3 with python3.  could you let me know how could you manage to do that? @yarbcy. I suspect you are still with greentea 1.7.2, because if you have 1.7.3 installed, the issue should be gone.

@mark-edgeworth could we have a greentea release for python3 as well.

mark-edgeworth, 2019-11-13:
Sorry, don't know how to do that. @madchutney might be able to help here.

dustin-crossman, 2019-11-13:
@jamesbeyond  @yarbcy Just to clarify the problem is fixed on our end (I updated our tools by using this command: pip3 install git+https://github.com/ARMmbed/mbed-os-tools.git@master#subdirectory=packages/mbed-greentea/). The greentea 1.7.3 upload on pypi still needs to get fixed, however, since we don't want anyone else running into the same issue.

jamesbeyond, 2019-11-14:
Thanks for reporting this issue. Yes we still need to fix the PyPI for Python3, there is a commit in 1.7.3 is for fixing this.

0xc0170, 2019-11-25:
> The greentea 1.7.3 upload on pypi still needs to get fixed, however, since we don't want anyone else running into the same issue.

@ARMmbed/mbed-os-tools Has this been fixed?

mark-edgeworth, 2019-11-25:
Not that I am aware of.

loverdeg-ep, 2019-11-25:
@mark-edgeworth Wait so who owns greentea now?

Or has the name change to mbed-test officially happened?

mark-edgeworth, 2019-11-25:
@jamesbeyond owns Greentea and Icedtea

jamesbeyond, 2019-11-25:
Yes, @ARMmbed/mbed-os-test should tag me also

yarbcy, 2019-12-04:
**Current status:**

1.7.3 is unavailable.

**Workaround:**
pip3 install git+https://github.com/ARMmbed/mbed-os-tools.git@master#subdirectory=packages/mbed-greentea/

jamesbeyond, 2019-12-04:
the fix will be in the next greentea release, which will happen soon

jamesbeyond, 2019-12-06:
Hi @yarbcy 
we having a new release of mbed-greentea 1.7.4, https://pypi.org/project/mbed-greentea/
would you able to confirm this issue is fixed, and closing the issue?

0xc0170, 2020-01-03:
I'll close this as resolved. Please reopen with an update

closed 2020-01-03 by 0xc0170

loverdeg-ep, 2020-01-08:
@millerc-ep