ARMmbed/mbed-os #11801: cycfg_capsense.c/h is missing in GeneratedSource directory

linlingao, 2019-11-01:
### Description of defect
cycfg_capsense.c/h is missing in targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CY8CPROTO_062_4343W/COMPONENT_BSP_DESIGN_MODUS/GeneratedSource directory, although design.cycapsense is present.

In addition, I also found that if I use design.modus to generate the C code, the generated routing code is different from the checked in version. I suspect design.modus might be stale. Can you make sure the generated code is recreate-able using design.modus?

#### Target(s) affected by this defect ?
TARGET_CY8CPROTO_062_4343W

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?
N/A

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
Run ModusToolbox 2.0/device-configurator. Open design.modus in the repo, and check the generated code against what's checked in. You should observe the capsense code is missing, routing code is different.

linlingao, 2019-11-01:
@ARMmbed/team-cypress @maclobdell

0xc0170, 2019-11-25:
@ARMmbed/team-cypress Please review